embedded-graphics/embedded-graphics #234: Batching pixel rendering by rows and blocks

lupyuen, 2019-12-29:
Hi All: Thank you so much for making [embedded-graphics] happen! :-)

I spent only 2 weeks studying the code and figuring out how we can make [embedded-graphics] and [st7735-lcd] render faster on PineTime smart watch, here are my findings...

[Optimising PineTime’s Display Driver with Rust and Mynewt](https://medium.com/@ly.lee/optimising-pinetimes-display-driver-with-rust-and-mynewt-3ba269ea2f5c?source=friends_link&sk=4d2cbd2e6cd2343eed62d214814f7b81)

Seems if we batch the pixels into rows and blocks, we can get significant improvement in rendering times. Yes it takes a little more RAM, but that can be managed by tuning the Row Iterator and Block Iterator.

So that's my suggestion for improving [embedded-graphics]. Thanks again, you are truly awesome! :-)

jamwaffles, 2019-12-29:
Hi! I'm glad you're enjoying this crate! I read through the article and that's a really interesting way of optimising draw calls. Another, simpler approach I've seen is [here in the SSD1306 driver](https://github.com/jamwaffles/ssd1306/blob/master/src/interface/i2c.rs) where the pixel data is batched into chunks of 16 bytes, but still fills a bounding box as opposed to rows/blocks.

I definitely think there's room for e-g to provide some kind of interface for optimising draw calls, but I'm not sure what that should be right now. I originally provided [`SizedDrawing`](https://docs.rs/embedded-graphics/0.5.1/embedded_graphics/trait.SizedDrawing.html), however it was poorly thought out and removed in 0.6.0. I'd like to see what solutions the community comes up with (including the one described in your article) before committing to a first-party API in embedded-graphics itself. Thank you for the suggestion though!

lupyuen, 2019-12-29:
Cool! Will stay tuned for updates as I bundle my code into the PineTime firmware, which will also include the [druid UI framework](https://medium.com/@ly.lee/porting-druid-rust-widgets-to-pinetime-smart-watch-7e1d5a5d977a?source=friends_link&sk=09b153c68483f7fa9e63350efd167b07). Thanks! :-)

closed 2019-12-29 by lupyuen