embedded-graphics/embedded-graphics #236: Make lib.rs docs more complete

jamwaffles, 2019-12-31:
Taken from [this tweet](https://twitter.com/Rubberduck203/status/1212023503668551680):

> Speaking of. The docs on http://docs.rs don’t show how the `display` is initialized. I had to go to the project’s readme to find it.

The examples in `lib.rs` have the display initialisation lines hidden. The README has a `MockDisplay` created with a note instructing the reader to swap `MockDisplay` for their driver of choice. The lib.rs examples should do the same so the magic `display` variable doesn't just come from nowhere.

rfuest, 2020-01-10:
On the topic of `MockDisplay`:
This is currently hidden from the docs, but used in the README. That is confusing for someone who tries to understand the examples and looks for it in the docs.
Should we make `MockDisplay` public? It could be useful for writing tests in other crates that depend on e-g.

jamwaffles, 2020-01-10:
Sounds good to me 👍

commit 3ce44189 (2020-03-18): Show display initialisation in lib.rs

Closes #236
commit b99ec405 (2020-03-19): Show display initialisation in lib.rs

Closes #236
PR #275 (2020-03-19): Final 0.6 stuff
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've  [... cut]
commit d2fb0abf (2020-03-19): Final 0.6 stuff (#275)

* Show display initialisation in lib.rs

Closes #236

* Add doc comments to simulator demos

Closes #214

* Remove crap hello example, redo chaining example

* Remove offscreen "example"/test

* Change lib.rs comments

* Example comment tweaks

* Demo 24x32 font

* More example comment tweaks
closed 2020-03-19 by jamwaffles