embedded-graphics/embedded-graphics #241: Return type for DrawTarget methods

rfuest, 2020-01-13:
While I was looking at the latest changes in the `ssd1331` crate I noticed that it has a `Ssd1331::flush` method but doesn't implement `DrawTarget::flush`. `DrawTarget::flush`, like all operations in `DrawTarget` doesn't have a return type, so calling `Ssd1331::flush` would only be possible with panicing when an error occurs.

Should we add an associated type to the `DrawTarget` crate, that let's the driver implementations choose an appropriate return type for possibly failing draw operations?

jamwaffles, 2020-01-13:
Makes sense to me. Associated type of generic type on the `flush()` fn itself? It doesn't make much sense to me to have an associated type only used in one method on the trait.

rfuest, 2020-01-13:
Should we only allow `flush` to fail or all drawing operations? Because depending on the implementation some drivers might directly draw without requiring flushing. Or should drawing operations and flush have two different return types?

> Associated type of generic type on the `flush()` fn itself? It doesn't make much sense to me to have an associated type only used in one method on the trait.

I'm not quite sure I understand how this would work without an associated type. Can you provide an example?

jamwaffles, 2020-01-13:
> Should we only allow flush to fail or all drawing operations?

All I think.

> Or should drawing operations and flush have two different return types?

Just one IMO. Different errors can be contained in an enum like the SSD1306/1331 crates do with pin errors and SPI errors.

> I'm not quite sure I understand how this would work without an associated type. Can you provide an example?

This is voided by your above comment anyhow :)

This is a breaking change due to modifying the public interface, right? Also, what should the default associated type be? `()`?

rfuest, 2020-01-13:
> Just one IMO. Different errors can be contained in an enum like the SSD1306/1331 crates do with pin errors and SPI errors.

But for buffered displays, having two might make it easier to use. If drawing operations can't fail the user of the API doesn't have to deal with handling `Result`s everywhere.

> This is a breaking change due to modifying the public interface, right?

Yes.

> Also, what should the default associated type be? `()`?

Associated type defaults aren't stable yet, so driver implementers have to set unused types to `()` them self.

jamwaffles, 2020-01-13:
> If drawing operations can't fail the user of the API doesn't have to deal with handling Results everywhere.

That would be nice, but I'm not sure how you'd implement it. Are you suggesting something like this?

```rust
pub trait DrawTarget {
    type ReturnType;

    // ...
}

struct Display;

impl DrawTarget for Display {
    type ReturnType = Result<(), ()>;

    // ...
}
```

Or just `type ReturnType = [whatever]` and each method returns a `Result<(), Self::ReturnType>`?

jamwaffles, 2020-01-13:
I experimented a lil bit with a `Result<(), Self::Error>` return type which made `Drawable::draw()` spit out a warning. I don't think it makes sense to make `Drawable::draw()` fallible but happy to be proved wrong.

rfuest, 2020-01-13:
I was thinking about something like:

```rust
pub trait DrawTarget {
    type ReturnFlush;
    type ReturnDraw;
    
    fn flush(&mut self) -> Self::ReturnFlush;
    fn draw_something(&mut self, ...) -> Self::ReturnDraw;
    // ...
}
```
This way the driver would have full control and can return `()` or `Result<A, B>` (most of the time `Result<(), Err>`).

But I didn't consider the default implementations in the trait and that `Drawable` also needs to pass the returned value back in `draw`. Maybe this works somehow, but I thing this is something that's easier to try in code.

rfuest, 2020-01-13:
IMO it makes sense:
See https://github.com/yuri91/ili9341-rs/blob/1956825a8eab44a39847fc0fc0c276a6c207d6af/src/lib.rs#L349 for an example where the driver panics in draw.

jamwaffles, 2020-01-13:
Ite, I'll get a PR together :+1:

PR #243 (2020-01-17): Draw target error type
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc)
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Gets `Drawable` and `DrawTarget` returning `Result`s allowing them to fail grac [... cut]
closed 2020-01-17 by jamwaffles