embedded-graphics/embedded-graphics #248: Consuming builders for styles

rfuest, 2020-02-04:
`PrimitiveStyleBuilder` and `TextStyleBuilder` are currently implemented as "Non-consuming builders" (https://doc.rust-lang.org/1.0.0/style/ownership/builders.html). But it might be beneficial to change them to consuming builders.

Consuming builders can be used to generate `const`s, which would be pretty useful IMO. This won't be a advantage immediately, because trait bounds aren't yet supported for `const fn`s, but this way this won't be a breaking change later.

It could also generate more efficient code, because `build` doesn't need to `clone` the style anymore. (this is just an assumption and not tested)

Are there any advantages of leaving the builders as non-consuming variants?

jamwaffles, 2020-02-04:
> Are there any advantages of leaving the builders as non-consuming variants?

I was going to type "giving multiple primitives the same style" but the code could of course just use the builder once and `.clone()` (or `Copy`) the resulting style. I can't think of any reason to make them non-consuming. I think it's a pretty common assumption that builders are of the consuming type anyway (citation needed) so that works out as well.

Should this go into 0.6.0 as a breaking change do you think?

rfuest, 2020-02-05:
> I was going to type "giving multiple primitives the same style" but the code could of course just use the builder once and .clone() (or Copy) the resulting style.

I did also consider creating multiple similar styles as a possible advantage of the non consuming builder. But you can also `clone` the builder before calling `build` to reuse to builder.

> Should this go into 0.6.0 as a breaking change do you think?

Yes, they were only introduced in the latest alpha and not many projects will already use them, so I think it's best to change them now.

jamwaffles, 2020-02-05:
Sounds good to me. I'll get a PR ready after #247 is merged. I cba with all the merge conflicts to do it in parallel :P

PR #256 (2020-02-17): Make builders consuming
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once [... cut]
closed 2020-02-17 by jamwaffles