embedded-graphics/embedded-graphics #249: how to compile a simulator code  ?

wagnerf42, 2020-02-05:
hi, I'm trying to compile a simulator example outside of its repository. (I'm a new user)
for example, I create a new project using cargo new and copy simulator/examples/fonts.rs into main.rs

note that compiling the example directly in the cloned simulator repository works fine.

the problem is I'm not too sure at what I should put into my Cargo.toml file as dependencies.
embedded-graphics="*"
embedded-graphics-simulator="*"
does not seem to work (it seems too old, the api seems to have changed)

if instead I put:
[dependencies.embedded-graphics]
version = "0.6.0-alpha.3"
features = [ "bmp", "tga" ]

[dependencies.embedded-graphics-simulator]
version = "0.2.0-alpha.1"

then the embedded-graphics-simulator dependency does not compile.
I get : 

   Compiling embedded-graphics-simulator v0.2.0-alpha.1
error[E0432]: unresolved import `embedded_graphics::Drawing`
  --> /home/wagnerf/.cargo/registry/src/github.com-1ecc6299db9ec823/embedded-graphics-simulator-0.2.0-alpha.1/src/lib.rs:83:5
   |
83 | use embedded_graphics::Drawing;
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

I guess it is because a new version is coming but I'm posting just in case there is an easy fix.

thanks

jamwaffles, 2020-02-05:
Hey, thanks for opening this. There are a lot of versions and breaking changes lying around at the moment, so sorry about that!

It's pretty grim, but until the non-alpha versions of crates are released and everything is fixed, you can pin crates to a specific commit by adding this to your `Cargo.toml`:

```toml
[patch.crates-io]
embedded-graphics = { git = "https://github.com/jamwaffles/embedded-graphics.git", rev = "4fc729a33b63f203ca059594aedf8463cc91da09" }
embedded-graphics-simulator = { git = "https://github.com/jamwaffles/embedded-graphics.git", rev = "4fc729a33b63f203ca059594aedf8463cc91da09" }
tinybmp = { git = "https://github.com/jamwaffles/embedded-graphics.git", rev = "4fc729a33b63f203ca059594aedf8463cc91da09" }
tinytga = { git = "https://github.com/jamwaffles/embedded-graphics.git", rev = "4fc729a33b63f203ca059594aedf8463cc91da09" }

[dependencies]
embedded-graphics = "=0.6.0-alpha.3"
embedded-graphics-simulator = "=0.2.0-alpha.1"
```

This pins to the latest commit at time of writing. Can you try that and see if it works?

wagnerf42, 2020-02-05:
it works, that's great !
thanks a lot. it seems like it's going to be a lot of fun to play with your crate.

jamwaffles, 2020-02-05:
> it works, that's great !

Awesome! I'll leave this issue open in case other people have the same problem.

> it seems like it's going to be a lot of fun to play with your crate.

Hope so ;). Enjoy!

jamwaffles, 2020-03-18:
Note to self: this should be closed when 0.6.0 proper is released.

jamwaffles, 2020-03-21:
0.6.0 is now released, so the hack described in this issue should no longer be required. Closing.

closed 2020-03-21 by jamwaffles