embedded-graphics/embedded-graphics #250: A better(?) way of building styles?

jamwaffles, 2020-02-07:
Might have some major issues idk, just recording this here before I forget it. In the future, I can imagine the styles for primitives deviating quite a bit from what we have now. The primary example that comes to mind is rectangle corner radii - they're not applicable to anything else, so doesn't make much sense to put them on `PrimitiveStyle`. Instead, we could go for styles unique to each primitive (with overlap for things like fill colour, stroke, etc) with something like the following:

```rust

trait StyleBuilder {
    type Style;

    fn builder() -> Self;   
    fn build(self) -> Self::Style;
}

struct Rectangle {
    // ...
}

#[derive(Default)]
struct RectangleStyle{
    corner_radius: Option<u8>,
    border_width: Option<u8>,
    // ...
}

struct RectangleStyleBuilder(RectangleStyle);

impl RectangleStyleBuilder {
    pub fn corner_radius(&mut self, rad: u8) -> &mut Self {
        self.0.corner_radius = Some(rad);
        self
    }

    pub fn border_width(&mut self, w: u8) -> &mut Self {
        self.0.border_width = Some(w);
        self
    }
}

impl StyleBuilder for RectangleStyleBuilder {
    type Style = RectangleStyle;

    fn builder() -> Self {
        Self(RectangleStyle::default())
    }

    fn build(self) -> RectangleStyle {
        self.0
    }
}

trait Primitive: Sized {
    type Style;

    fn into_styled(self, style: Self::Style) -> Styled<Self, Self::Style> {
        Styled::new(self, style)
    }
}

impl Primitive for Rectangle {
    type Style = RectangleStyle;
}
```

This could then be used like this (untested):

```rust
Rectangle::new(/* ... */)
    .into_styled(
        RectangleStyleBuilder::builder().corner_radius(10).build()
    )
    .draw(&mut display);
```

Thoughts?

rfuest, 2020-02-07:
I haven't looked at the code yes, but I have a comment about rounded rectangles.
There are different ways of looking at rounded rectangles. Looking at this from a HTML/CSS perspective the corner radius is a style property. But it could also be considered to be a property of the primitive or probably a special `RoundedRectangle` primitive. This would make sense if primitives could be used for other applications (see #156) IMO.

jamwaffles, 2020-02-07:
> This would make sense if primitives could be used for other applications (see #156) IMO.

Hm, yes it would. Performing hit tests on a rounded button in a UI or something like that would need to make use of the non-styled rect.

So, that was a poor example. How about line cap styles, or the fill mode of a non-convex polygon, etc. I think there's definitely a place for primitive-specific styles, even if my example given above sucks :grin:

rfuest, 2020-02-07:
If each primitive has a specific style type how would I draw different primitives with, for example, the same stroke?

jamwaffles, 2020-02-08:
Hm, interesting. I suppose there could be a `BaseStyle` struct that the other styles can `impl From<>` for? Doesn't feel very pleasant though.

closed 2020-04-13 by jamwaffles