embedded-graphics/embedded-graphics #263: DrawTarget::size is broken

rfuest, 2020-02-26:
The recent changes in #261 caused some issues I didn't noticed because apparently the `size` function isn't used anywhere. But when I tried to refactor #259 an error occurred:
```
error[E0283]: type annotations needed
  --> simulator/examples/hello-world.rs:23:48
   |
23 |     let bottom_right = Point::zero() + display.size() - Point::new(1, 1);
   |                                                ^^^^ cannot infer type for type parameter `PixelC`
   |
   = note: cannot resolve `_: embedded_graphics::pixelcolor::PixelColor`
   = note: required because of the requirements on the impl of `embedded_graphics::draw_target::DrawTarget<_>` for `embedded_graphics_simulator::display::SimulatorDisplay<embedded_graphics::pixelcolor::binary_color::BinaryColor
```
That makes sense, because `size` is the only function that doesn't have any relation to the color type. I'm not sure what the best way to fix this is. Some options are:
* Remove automatic color conversion
  This would make using different color format images harder to use on color displays. But maybe some kind of conversion function (maybe as an extension trait) could be added to make this simpler. I guess at the moment you would need to use `map` on the iter to convert the colors manually, which isn't comfortable at all.
* Change the `draw_...` functions in the `DrawTarget` trait
   ```rust
   /// Draws a pixel on the display.
   fn draw_pixel(&mut self, item: drawable::Pixel<C>) -> Result<(), Self::Error>;
   ```
   to:
   ```rust
   /// Draws a pixel on the display.
   fn draw_pixel<PixelC>(&mut self, item: drawable::Pixel<PixelC>) -> Result<(), Self::Error>
      where PixelC: PixelColor + Into<C>;
   ```
   But then `C` should be a associated type instead, that represents the native color type of the framebuffer.
* Remove `size` from `DrawTarget`
  This way all functions would depend on `C` and `size` could be moved to something like `DrawTargetDimensions`.

Having to change this now isn't ideal, but I guess we need to do something, because all drivers will depend on this.

jamwaffles, 2020-02-27:
Isn't that because the pixel color type isn't defined [here](https://github.com/jamwaffles/embedded-graphics/pull/259/files#diff-a8c32b73fb30b81a0aa98efb1a3fdeefR12)? One or more of the SSD1306 examples use `disp.size()` without issue, so I think this might be ok, at least from the display driver side? I think this might be more to do with the simulator being a slightly odd case.

rfuest, 2020-02-27:
Because the SSD1306 is a monochrome display it implements `DrawTarget<BinaryColor>` which works fine.

The issue only exists for color displays that implement `DrawTarget<C> where C: PixelColor + Into<FramebufferC>`. If a color display implements `DrawTarget<FramebufferC>` instead `size()` works but working with images that aren't stored in the same color type becomes harder. This is especially true now that `Image`s are handled by a separate function in `DrawTarget` which makes it impossible to just use `map` to convert the colors coming from an image iterator without loosing hardware acceleration.

jamwaffles, 2020-02-28:
I just encountered this migrating the SSD1331 driver which uses `Rgb565` so I see what you mean now.

I'm in favour of adding an associated type and changing all the draw methods. No published drivers on crates.io use beta.1 yet so I think we're ok to add a breaking change.

jamwaffles, 2020-02-28:
Actually maybe not. I think that approach would require major changes to `Drawable` too?

jamwaffles, 2020-02-28:
I found a workaround/kludge that does work, although it's not very clean: https://github.com/jamwaffles/ssd1331/pull/5/files#diff-04c6e90faac2675aa89e2176d2eec7d8R88-R103

Let me know what you think. A breaking change can be released as a last resort for sure, but if it's hackable then maybe it's best to leave it.

rfuest, 2020-02-28:
> I found a workaround/kludge that does work, although it's not very clean: https://github.com/jamwaffles/ssd1331/pull/5/files#diff-04c6e90faac2675aa89e2176d2eec7d8R88-R103

That works and I would be OK to keep this, if it would only affect internal code, but expecting the user of the API to write this to access the display size doesn't seem like a good idea IMO.

I did think about this some more and would suggest the following solution for 0.6: We change `DrawTarget` to use an associated type, but only change the `draw_image` method to use `Into<FramebufferC>` as a color type. The styled primitives created for drawing onto the `DrawTarget` are most likely already using the correct color type, so that changing all drawing methods won't have much benefit IMO. But requiring all images to use the same color type as the display would be impractical and changing the only recently introduced `draw_image` shouldn't break many applications.

I'm not quite sure if the automatic color conversion for images is a good idea in general or we should prefer explicit conversion at that point. But IMO this is the best solution for 0.6, that doesn't break to much code or requires more extensive changes to e-g.

jamwaffles, 2020-02-28:
> We change DrawTarget to use an associated type

Sure, let's go for that then.

> but only change the draw_image method to use Into<FramebufferC> as a color type

We might as well change everything in one go, right? I imagine it would be beneficial to change all the other `draw_*` methods for consistency, no? This is going to be a breaking change with shifting to an associated type anyway, so might as well go all out.

> I'm not quite sure if the automatic color conversion for images is a good idea in general or we should prefer explicit conversion at that point

Automatic is fine IMO. If a specific implementation is required, the programmer can always call `.into_iter().map(...)` to set up a custom colour mapping.

jamwaffles, 2020-02-28:
Do you have time to take your suggested changes on or shall I carve out some minutes?

rfuest, 2020-02-28:
> Automatic is fine IMO. If a specific implementation is required, the programmer can always call `.into_iter().map(...)` to set up a custom colour mapping.

Sure, but that will break hardware acceleration.

> We might as well change everything in one go, right? I imagine it would be beneficial to change all the other `draw_*` methods for consistency, no? This is going to be a breaking change with shifting to an associated type anyway, so might as well go all out.

I would prefer to not change this for 0.6. There will be changes to the `DrawTarget` trait in 0.7, which could make it necessary to reconsider how color conversion is handled. If we add the automatic conversions now it might be harder to implement the necessary changes without breaking too much code. And if if turns out to be a good idea the conversion could still be added in 0.7, which should cause minimal code breakage, because most implementation of `DrawTarget` will use the default implementations of the `draw_...` functions.

rfuest, 2020-02-28:
> Do you have time to take your suggested changes on or shall I carve out some minutes?

Yes, I can prepare a PR for this.

rfuest, 2020-02-28:
Hmm, I just tried to implement this and this doesn't work as expected:
Let's imagine that you try to draw an image of type `Rgb565` to a `DrawTarget<Rgb888>`. The image implements `Drawable<Rgb565>` and the `draw` method requires a `DrawTarget<Rgb565>`. So drawing the image using `image.draw(display)` wouldn't work, even if `DrawTarget::draw_image` supports color conversion.

Should we move this entire change to 0.7? Any attempt to fix this properly would cause pretty big changes to the API. If the display drivers for color displays implement `DrawTarget<RgbXXX>` instead of some generic type most features would work. Hardware accelerated drawing of images would at least work for equal color types, which wouldn't be ideal but an improvement over pervious versions. Other images could still be drawn without hardware acceleration using something like `image.into_iter().map(|Pixel(p, c)| Pixel(p, c.into())).draw(&mut display)` until something better is implemented in 0.7.

jamwaffles, 2020-02-28:
Yeah this doesn't seem like a simple problem to fix. Let's move it to 0.7.

closed 2020-06-04 by jamwaffles