embedded-graphics/embedded-graphics #268: Variable width font styling

jamwaffles, 2020-03-11:
Off the back of #267, I think it should be possible to make all built-in fonts variable width by default. Here's my idea:

* As far as I can see, all built in fonts can support variable widths, so set the `VARIABLE_WIDTH` property on `Font` to `true` by default (but retain it so there's an option to enforce monospace behaviour for third party fonts)

* One of the following ways of specifying letter spacing
    1. Two properties, `letter_spacing` for the number of pixels between characters (default 1 px) and `font_family` (inspired by the CSS property of the same name - could use a better one) which is an enum of `Monospace` or `VariableWidth`.

    2. An enum that encapsulates both properties:

        ```rust
        enum LetterSpacing {
            Monospace,
            VariableWidth { spacing: u32 }
        }
        ```

    3. Other suggestions accepted!

* If a font specifies `VARIABLE_WIDTH: false` in its impl of `Font`, the letter spacing properties on `TextStyle` shall have no effect.

    * _Or_ it could only respect the letter spacing but not shrink spacing greater than 1px?

@rfuest any thoughts?

aurelj, 2020-03-11:
I like the proposition overall.

But there is one catch. Currently the variable width font code only remove the blank columns at the right of the characters but not on the left, and the current `Monospace` fonts have characters such as `i` or `!` smack in the middle. So the variable width font code should first be improved to also remove blank columns at the left of the characters before being able to use every font as variable width.

jamwaffles, 2020-03-11:
> But there is one catch.

Ah, hm that's a good point. It shouldn't be too difficult to add a left-trimming iterator too though? I'd prefer not to modify the font bitmaps as that would complicate/break existing monospace behaviour.

I think a longer term goal might be to adopt the BDF font format (or a custom derived format for space reasons) which IIRC allows defining a bounding box for a character inside the nominal grid.

#269 also raised the ability to add line spacing now that multiline text is supported from that PR. `TextStyle` should probably also be able to set line spacing (but only _between_ lines).

rfuest, 2020-03-11:
I'm not a fan of trying to automatically crop glyphs to convert a monospace font into a proportional one. This isn't very flexible and doesn't allow any kind of kerning.
And IMO selecting if a font is monospaced or proportional isn't something that belongs into `TextStyle` but inside the `Font`.

My suggestion:
* Don't automatically crop glyphs, but instead add an additional function to the `Font` trait to determine the width of a glyph. This way each `Font` impl can choose any method of determining the width of a glyph, like using a table of glyph width or also the current way of removing empty columns.
* If the same font data looks good as a monospaced and a proportional variant, provide two `Font` impls instead: e.g. `Mono6x8` and `Prop6x8`.

jamwaffles, 2020-11-16:
Does #470 close this?

rfuest, 2020-11-17:
It does.

closed 2020-11-17 by rfuest