embedded-graphics/embedded-graphics #272: Possible Enhancement: Make DrawTargets Debuggable

caemor, 2020-03-17:
- Version of embedded-graphics in use (if applicable): 0.6-beta2
- Target: [simulator] 

## Description of the problem/feature request/other

Make DrawTarget a Supertrait of Debug to be able to debug/print/log/... the new error type.

## Test case (if applicable)

```rust
use embedded_graphics::pixelcolor::BinaryColor::{Off as White, On as Black};
use embedded_graphics::{
    drawable::Drawable,
    fonts::*,
    geometry::Point,
    pixelcolor::BinaryColor,
    style::{TextStyle, TextStyleBuilder},
    text_style, DrawTarget,
};
use log::*;

pub fn draw_text<T: DrawTarget<BinaryColor>, F: Copy + Font>(
    display: &mut T,
    text: &str,
    top_left: Point,
    font: F,
) {
    if let Err(e) = Text::new(text, top_left)
        .into_styled(text_style!(
            font = font,
            text_color = Black,
            background_color = White
        ))
        .draw(display)
    {
        //println!("Error: {:?}", e);
        //log::error!("Error: {:?}", e)
        //write!("Error: {:?}", e);
        //...
    }
}
```

rfuest, 2020-03-17:
Maybe I'm missing something, but why does `DrawTarget` need to be a supertrait of `Debug` to be able to output an error? Isn't it enough when `DrawTarget::Error` implements `Debug`?

caemor, 2020-03-17:
No, I think you are right

rfuest, 2020-03-17:
IMO adding a `Debug` trait bound to `Error` shouldn't be necessary either. In your example you could add `where T::Error: Debug` to be able to output the error for a generic `DrawTarget`. Or do you have a special use case in mind where the trait bound in e-g itself is required?

caemor, 2020-03-17:
Thanks thats already enough :-)

closed 2020-03-17 by caemor