embedded-graphics/embedded-graphics #273: Possible Enhancement: Set default stroke_width of PrimitiveStyle to 1

caemor, 2020-03-17:
- Version of embedded-graphics in use (if applicable): 0.6-beta2
- Target: [simulator]

## Description of the problem/feature request/other

Currently the default stroke_width is zero. This at least for lines and triangles leads to at first unexpected results (see example from the eg_lines description)

### Proposal:

Set the Default to 1 so lines, circles, rectangles and triangles are drawn by default (not the filling!).

### Another alternative Proposal:

Make 'stroke_color' a more complex type with a combined width and let it set width to 1 by default and therefore make it only usable for width > 1.

## Test case (if applicable)
 
```rust
 use embedded_graphics::{
     egline,
     pixelcolor::Rgb565,
     prelude::*,
     primitive_style,
     primitives::Line,
     style::{PrimitiveStyle, Styled},
 };

 let line: Styled<Line, PrimitiveStyle<Rgb565>> = egline!(start = (10, 20), end = (30, 40));

 let stroke_line: Styled<Line, PrimitiveStyle<Rgb565>> = egline!(
     start = (10, 20),
     end = (30, 40),
     style = primitive_style!(stroke_color = Rgb565::BLUE)
 );
```

rfuest, 2020-03-17:
The default value of `stroke_width` was recently changed to `0`, so there might be still some docs that don't reflect this change. The reason for this change was to not have any arbitrary default value for stroke width and require setting the value explicitly in the application. The `PrimitiveStyle::with_stroke(color, width)` uses a mandatory width argument to make sure the width is always specified.

> Make 'stroke_color' a more complex type with a combined width and let it set width to 1 by default and therefore make it only usable for width > 1.

That might be a good idea, something like this could work:
```rust
enum Stroke<C: PixelColor> {
    None,
    Some { color: C, width: NonZeroU32 },
}
```
But this would make it harder to modify an existing style, because you couldn't simply call `style.color = Rgb888::Red;` anymore.

caemor, 2020-03-17:
Thanks for the quick response. I am just using the with_stroke alternative then :-) :+1:

closed 2020-03-17 by caemor