embedded-graphics/embedded-graphics #281: General sprite support

jamwaffles, 2020-03-21:
Fonts are "just" spritemaps. Should e-g have builtin support for sprites that can handle animations and other standard sprite needs, with the fonts migrated to use that system? The alternative is to let that be built in an external crate, but I don't think there'd be that much work to adapt the existing font system to general sprite support.

rfuest, 2020-03-21:
What do you mean by "handle animations" and "other standard sprite needs"? Does this mean that you want to add a whole animation system with easing etc. to e-g?

jamwaffles, 2020-03-21:
My bad, wrote that in a bit of a rush. I mean things like getting a sprite layer by index and, as I don't know much about them, other foundational things that sprite systems should usually have. Animations, etc should be handled in another layer. My idea is to take the current fonts system and slightly generalise it so it can be used for other types of tiled pixel data.

rfuest, 2020-03-21:
I was thinking about adding support for sub images to the image module, or at least the built in raw image support A sub image would be a reference to some rectangular part of some image data. This would essentially be sprites without transparency and a first step towards sprite support.

The same code could also be reused for font rendering, which would reduce code duplication and allow fonts with more than one bit per pixel.

jamwaffles, 2020-03-21:
Yeah, that's the kind of thing I was going for. Transparency is an interesting point. Perhaps an associated const (or other way of specifying) what colour should be treated as transparent? For example, to treat black as transparent:

```rust
trait Sprite<C: PixelColor> {
    const SUB_SIZE: Size;

    const TRANSPARENT: C;

    // etc
}

struct BlackTransparentSprite;

impl Sprite<Rgb565> for BlackTransparentSprite {
    const SUB_SIZE = Size::new(10, 10),

    const TRANSPARENT = Some(Rgb565::BLACK)

    // etc
}
```

Ignore all the specifics and names and stuff haha, but maybe the associated const idea could work.

This also raises the question of transparency support generally :thinking:.

rfuest, 2020-03-21:
Using an associated constant for the sprite size seems limiting. And requiring one type per sprite size could would probably increased binary sizes because of monomorphization.

As a first step I would go with something like this:

```rust
struct SubImage<'a, I> {
   image: &'a I,
   offset: Point,
   size: Size,
}

let sprite_sheet = Image::new(128, 128, DATA);
let player_sprite = sprite_sheet.get_sub_image(Point::zero(), Size::new(32, 32));

Image::new(&player_sprite, player_position).draw(&mut display)?;
```

> This also raises the question of transparency support generally.

This is definitively something we need to work on at some point. Adding support for binary transparency would be a good start IMO, because we don't have to care about alpha blending.

The `SubImage` struct above doesn't support any transparency by itself, which has the advantage that it can be guaranteed that the iterator iterates over exactly `size.width * size.height` pixels and can support hardware acceleration. Drawing such a sprite with some transparent pixels is pretty easy using `filter`:
```rust
Image::new(&player_sprite, player_position)
    .into_iter()
    .filter(|Pixel(_, c)| c != Rgb565::MAGENTA)
    .draw(&mut display)?;
```
This manual code could be moved into a wrapper around an image:
```rust
struct BinaryTransparencyImage<I, C> {
    image: I,
    transparent: C,
}

let sprite_sheet = BinaryTransparencyImage::new(Image::new(128, 128, DATA), Rgb565::MAGENTA);
let player_sprite = sprite_sheet.get_sub_image(Point::zero(), Size::new(32, 32));
```

jamwaffles, 2020-03-22:
Ah, I think we have two different scenarios in mind:

1. Packed sprites of different sizes in a single image
2. Animation frames for a single sprite packed into a single image, all with the same size

I'd like to be able to support both cases. In the second case, all that's really needed is a "get by index" method and a way of specifying the sprite size. It could be a wrapper around the solution for the first point?

> Adding support for binary transparency would be a good start IMO, because we don't have to care about alpha blending.

Oof yeah, alpha blending will be annoying. It'd be a prerequisite for AA on primitives I think, so we'll have to tackle it at some point :grimacing:

rfuest, 2020-03-22:
> Ah, I think we have two different scenarios in mind:

Yes we did.

> It could be a wrapper around the solution for the first point?

It could, but maybe that should be handled in the application instead. An builtin mapping from index to a position in a sprite sheet would require a predefined layout of the sprites in the sprite sheet. And IMO referring to a sprite by `Point::new(index * SPRITE_WIDTH, 0)` instead of `index` isn't much more difficult. But this would make something like `Point::new(animation_frame * SPRITE_WIDTH, enemy_type * SPRITE_HEIGHT)` possible, if sprites are laid out in a 2D grid.

> Oof yeah, alpha blending will be annoying. It'd be a prerequisite for AA on primitives I think, so we'll have to tackle it at some point

Yeah, that will be fun :wink: And we will also need to take hardware acceleration into account, because some µC like the bigger STM32 support alpha blending in hardware.

jamwaffles, 2020-03-22:
> It could, but maybe that should be handled in the application instead.

Sure, let's go for that. I think this feature is morphing into a more general "get sub image from image", so let's go with that direction for an implementation.

rfuest, 2020-03-24:
Do you want to work on this or should I give it a try?

jamwaffles, 2020-03-24:
My plate is full of other stuff at the moment so feel free to take it.

jamwaffles, 2020-06-01:
Current name as suggested in https://github.com/jamwaffles/embedded-graphics/issues/355#issuecomment-637020147 is `SubImage`.

closed 2020-08-06 by rfuest