embedded-graphics/embedded-graphics #283: Add ellipse primitive

jamwaffles, 2020-03-22:
Similar to `Circle`, defined by top left as per #274, but with a width/height given as a `Size`. E.g.:

```rust
// Create an ellipse 10px wide, 20px tall
let ellipse = Ellipse::new(Point::zero(), Size::new(10, 20));

// Create a circle 20px in diameter
let ellipse = Ellipse::new(Point::zero(), Size::new(20, 20));
```

Questions:

1. [ ] Should `Circle` become a simple wrapping struct to `Ellipse`, or should it be removed entirely? IMO `Circle` should be kept in.

rfuest, 2020-03-22:
> Should `Circle` become a simple wrapping struct to `Ellipse`, or should it be removed entirely? IMO `Circle` should be kept in.

I think I agree with that.

jamwaffles, 2020-03-22:
Cool cool. Original comment updated.

aurelj, 2020-04-07:
While this definition of an `Ellipse` covers pretty much all the common use cases, it should be noted that it doesn't allow drawing ellipses with their major and minor axis not horizontal and vertical.

I don't know if this case is worth supporting, but it is at least good to clearly know the limitations of the chosen definition.

PR #320 (2020-04-14): Ellipse
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened t [... cut]
closed 2020-04-14 by jamwaffles