embedded-graphics/embedded-graphics #284: Add filled/unfilled arc primitive

jamwaffles, 2020-03-22:
Add an `Arc` primitive. An arc is defined by a start position, end position and center point. 

It may also be useful to add the ability to specify an arc by its center point, start/end number of degrees from 0 (which points to the top screen edge) and radius.

```rust
// Create the top-left quadrant of an ellipse
// (center, start, end)
let arc = Arc::new(Point::new(20, 20), Point::new(5, 20), Point::new(20, 10));

// Create a circular segment in the top-right quadrant
// (center, start angle, end angle, radius)
let segment = Arc::from_degrees(Point::new(20, 20), 0, 90, 10)
```

rfuest, 2020-03-22:
Should filled and unfilled arcs use the same primitive? In #77 we decided to use different types for filled and unfilled polygons. So we might want to do the same for arcs.

There are two common variants of how a filled arcs are handles. One results in a pie shape and the other just connects the two arc ends with a straight line:
[image]
Should we support both?

jamwaffles, 2020-03-22:
> Should filled and unfilled arcs use the same primitive?

In #77, I think we're going to disallow fills on `Polyline`s which is why they're separate from `Polygon`s, whereas here an `Arc` has an optional fill, so I think filled and unfilled arcs should probably be the same type. That said, if we use the same type for `Arc`s, maybe the `Polyline`/`Polygon` distinction should be reduced into a single `Polygon` type too?

> Should we support both?

Hmm. Yes, ideally, although if I had to choose one it'd be the "pie" still fill. If we can do both, how would you suggest choosing the fill style?

rfuest, 2020-03-22:
> maybe the `Polyline`/`Polygon` distinction should be reduced into a single `Polygon` type too?

I would like to keep them distinct, because a polygon encloses an area an can, for example, be used for hit testing, which doesn't make sense for an open path like a polyline

> Hmm. Yes, ideally, although if I had to choose one it'd be the "pie" still fill. If we can do both, how would you suggest choosing the fill style?

I don't know yet. Setting the fill style in `PrimitiveStyle` would work for drawing, but unstyled `Arc`s couldn't be used for hit testing anymore. One solution would be to use distinct `Arc`, `Segment` and `Sector` primitives.

IMO it should also be possible to set if the stroke is also applied to the linear segments.

jamwaffles, 2020-03-23:
> I would like to keep them distinct, because a polygon encloses an area an can, for example, be used for hit testing, which doesn't make sense for an open path like a polyline

Makes sense :)

My target feature for opening this issue was to be able to create rounded joiners for polylines with a line width greater than 1px. This would need the ability to create a filled sectors, which may be better off in a separate issue. Shall I:

1. Reduce the scope of this issue to _unfilled_ arcs only and
2. Create a new issue for a `Sector` primitive which supports both stroke and fill? 

If I understand correctly, a `Sector` would have a fill behaviour like the left example in your image above.

> IMO it should also be possible to set if the stroke is also applied to the linear segments.

Not relevant to my initial reason, but yes that would be a useful addition, although that feature could certainly be added later. This property would be specified on `PrimitiveStyle`, right?

rfuest, 2020-03-23:
`Arc` and `Sector` will be very similar in some aspects, like the constructors or the fields in the structs, so it might be a good idea to add them at the same time. And adding the conversion functions between `Arc`s and `Sector`s will also require both types to be implemented. `Arc`s and `Sector`s could also be added as primitives only at first and drawing could be implemented later. But if you think it's easier to keep track of this with to separate issues you can also split them like you described.

> If I understand correctly, a `Sector` would have a fill behaviour like the left example in your image above.

Exactly.

> Not relevant to my initial reason, but yes that would be a useful addition, although that feature could certainly be added later. This property would be specified on `PrimitiveStyle`, right?

On second thought we might not need this. IMO a stroke on `Sector` and `Segment` should always contain the linear and arc segments. If someone wants to draw a filled `Sector` or `Segment` where the stroke only applies to to arc it can be split into two drawing operations: First fill the `Sector` without stroke and then draw an `Arc`.

jamwaffles, 2020-03-30:
> But if you think it's easier to keep track of this with to separate issues you can also split them like you described.

You make good points. I'll keep this as one issue and implement everything at the same time. I like the idea of being able to convert between them as well.

> First fill the Sector without stroke and then draw an Arc.

That works :) 

For sectors, I think we also need a way of specifying whether it's a major or minor sector:

[image]

I also dredged up #83 which would be another candidate for arc and sector support.

rfuest, 2020-03-30:
> For sectors, I think we also need a way of specifying whether it's a major or minor sector:

That's a good point and not only relevant for sectors but all arcs.

For arcs defined by angles there are two methods if encountered before:

One is using separate functions for clockwise and counter clockwise arcs (like [`cario-arc`](https://www.cairographics.org/manual/cairo-Paths.html#cairo-arc) and `cairo-arc-negative` in cairo and the optional [`anticlockwise`](https://developer.mozilla.org/en-US/docs/Web/API/CanvasRenderingContext2D/arc) argument in the HTML5 canvas API).

The other is using a start angle and sweep angle instead of a start and end angle. The sign of the sweep angle defines the direction of the arc.

jamwaffles, 2020-03-30:
I think my preference would be towards a start and sweep angle as there is no switch (like the `anticlockwise` arg) and only one function required. What would be your preference? I think the start and sweep angles would have to be `f32` to get the accuracy required. Of course an arc defined by points could just use the integers provided by `Point`.

rfuest, 2020-03-31:
> I think my preference would be towards a start and sweep angle as there is no switch (like the `anticlockwise` arg) and only one function required. What would be your preference?

That sounds like a good solution.

> I think the start and sweep angles would have to be `f32` to get the accuracy required.

For most smaller displays integer degrees would provide enough resolution. A circle with 128 pixel diameter has a circumference of ~400 pixels and a single degree step would be about 1.1 pixel. But maybe it's better to also think about larger displays and antialiased drawing which could benefit from the additional resolution.

jamwaffles, 2020-04-06:
[This comment](https://github.com/jamwaffles/embedded-graphics/pull/310#issuecomment-609909504) raises an interesting point:

> But I have an issue with your `(center: Point, start: Point, end: Point)` definition. It looks over-constrained to me. What if the distance between center and start is not the same as the distance between center and stop ? We couldn't draw a Circle or Arc that passes thru both start and stop.

My plan for this was to draw a segment of an ellipse, where the start and end of the segment is perpendicular to the line from `center -> start` and `center -> end`. Should this be a thing that `Arc` and/or `Segment` can do?

aurelj, 2020-04-07:
Oh sure, `Segment` of ellipse should be supported. But if we keep `Ellipse` separate from `Circle` (as suggested by #283) we should also keep `Segment` of ellipse separate from `Arc` of circle.

And I guess that the cases were you really want a `Segment` of ellipse rather than an `Arc` of circle are very uncommon.

jamwaffles, 2020-06-11:
Closed by #310

closed 2020-06-11 by jamwaffles