embedded-graphics/embedded-graphics #288: 0.6.x releases in tandem with 0.7.x work

jamwaffles, 2020-03-26:
Before we get too far ahead, there will be fixes and improvements that can be released as part of the 0.6.x series whilst 0.7.x breaking changes happen in parallel. To this end, should we create a 0.7 feature branch to merge breaking PRs into? There are already breaking changes in master, so I'd have to branch off an earlier commit and force push master. Is there a better/preferred way?

rfuest, 2020-03-27:
Force pushing master seems like a bad idea IMO. For now I would suggest to keep doing 0.7 development in master and create an 0.6 branch for 0.6 patch releases. We can reconsider changing the way we use branches after 0.7 is released.

jamwaffles, 2020-03-27:
Sure, that makes sense. Cherry picking stuff from master into 0.6 should work ok.

jamwaffles, 2020-03-27:
On second thoughts, I would prefer to keep releasing from master, and work in a 0.7 feature branch. I think we'll run into problems with git tags and `cargo-release` and probably other stuff later if we don't.

Splitting off 0.7.0 into a feature branch would require one force push to move the `Circle` changes in master, which I think is acceptable. Force pushing isn't great, but the benefits outweigh the drawbacks this time I think?

rfuest, 2020-03-27:
I haven't used `cargo-release` before, so you can better judge if this will cause any problems.

We could use `git revert` to add a new commit that removes the circle changes instead of force pushing. IMO that would be a cleaner solution.

jamwaffles, 2020-03-27:
Ah, yeah a revert would work. Let me rustle a PR up.

rfuest, 2020-03-27:
Can you also create the development branch and push the circle changes to that branch? This would allow me to finish the rectangle PR.

jamwaffles, 2020-03-27:
Yep, I just pushed the [`v0.7.0`](https://github.com/jamwaffles/embedded-graphics/tree/v0.7.0) branch. I reverted the revert for the circle PR on it which is messy but should get us back to the correct state and (hopefully) make it easy to rebase master onto v0.7.0 to keep everything in sync.

jamwaffles, 2020-03-27:
Do we want to create a long lived PR for that branch or just leave it in git for now? Might useful for us and other watchers to see what's going to be in 0.7.0 if we do create it.

rfuest, 2020-04-03:
The `0.7` branch is behind master and it's easy to forget to merge changes in `master` to the development branch. IMO committing all changes to a development branch and cherry picking relevant changes to `0.6` would work better. By doing all the work in `master` and using one branch per release, it would also be possible to maintain more than one previous version, because every major release would have a separate branch.

But I'm not very experienced with the different git workflows and maybe there are better methods. I just think that the current one isn't working great.

rfuest, 2020-04-04:
I meant:
* Delete the 0.7.0 branch.
* All PRs and changes are merged into master.
* When a new stable version is released a permanent branch is created (e.g. `0.6`)
* Master is always one version ahead of the last stable branch (at the moment `0.7`)
* All non breaking changes in master can be cherry picked into the stable branch

> I'd like to be able to clean up these branches but keep the tag but I don't think that's possible, right? They'd have to live forever once created if we don't want to break links.

If the stable version branches are permanent they could be used for tags.

jamwaffles, 2020-04-04:
Ok great, I think we're aligned :). Next steps:

1. I'll create a PR to un-revert the circle changes in #274  back into master (this is the only change not in master currently)
2. Once that's merged, I'll then delete the `v0.7.0` branch and we can continue on master as usual

Anything I've missed here?

rfuest, 2020-04-04:
If we want to release more `0.6.x` releases it might be a good idea to create the `0.6` branch before we merge the circle changes into master.

jamwaffles, 2020-04-04:
That's true. Just for completeness' sake:

1. I'll create the `v0.6` branch
1. I'll create a PR to un-revert the circle changes in #274  back into `master` (this is the only change not in master currently)
1. Once that's merged, I'll then delete the `v0.7.0` branch and we can continue on master as usual
1. Any non-breaking changes for 0.6 can be cherry picked from `master` into `v0.6`.

rfuest, 2020-04-04:
SGTM :+1:

rfuest, 2020-04-04:
How should we handle changelogs in the future? If some changes are cherry picked into a stable branch, they will be added to the changelog for that version. Should we manually merge these additions to the changelog back to `master` after each minor release or handle this once before a new stable version is released. Or maybe something else?

jamwaffles, 2020-04-04:
Yeah, I think a merge from the release branch into `master` would have to be it. I can't think of a better way, so let's see how it works when a 0.6 release happens? We can tune the process from anything learned then.

jamwaffles, 2020-04-04:
`v0.7.0` is now deleted. We should be good going forward. I think this now unblocks #286?

jamwaffles, 2020-04-04:
Also, please close this issue if you think it should be.

rfuest, 2020-04-04:
Yes, once the naming issues are resolved #286 can be merged.

closed 2020-04-04 by rfuest