embedded-graphics/embedded-graphics #289: Draw Gray4 image on Rgb565 display

fkohlgrueber, 2020-03-27:
- Version of embedded-graphics in use (if applicable): 0.6
- Target: real hardware
- If using real hardware: ST7735 connected to 1,8" RGB565 display

## Description of the problem

I'm trying to display a Gray4-encoded image on a Rgb565 display. I know that I need to convert from Gray4 to Rgb565 at some point, but don't know when and how. I could probably store the image encoded as Rgb565 in flash, but that seems to be wasting a lot of space. What's the recommended approach for this?

This is the code I'm currently using, which errors at the draw call due to different PixelColor types.

```
let mut disp = st7735_lcd::ST7735::new(spi, dc, rst, true, false, 160, 128);
...
let image_raw: ImageRawLE<Gray4> = ImageRaw::new(include_bytes!("..."), 128, 111);
let image: Image<_, Gray4> = Image::new(&image_raw, Point::new(0, 0));
image.draw(&mut disp).unwrap();  // Error
```

Thanks in advance!

rfuest, 2020-03-27:
Currently images with different color types can only be drawn by converting them into an iterator and using `map` to convert the colors:
```rust
let image_raw: ImageRaw<Gray4> = ImageRaw::new(include_bytes!("...", 128, 111);
let image = Image::new(&image_raw, Point::zero());
image
    .into_iter()
    .map(|Pixel(p, c)| Pixel(p, c.into()))
    .draw(&mut disp)
    .unwrap();
```

fkohlgrueber, 2020-03-29:
Alright, thanks a lot!

closed 2020-03-29 by fkohlgrueber