embedded-graphics/embedded-graphics #294: Please make sdl dependency optional

Boscop, 2020-03-28:
Thanks for this great crate. I'm using `embedded-graphics-simulator` to render data into ImageBuffers and then into glium textures, for use with imgui-rs.
But currently I get linking errors because I don't have SDL:
```
  = note: libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.8.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_LoadLibrary referenced in function _ZN4sdl25video43_$LT$impl$u20$sdl2..sdl..VideoSubsystem$GT$27vulkan_load_library_default17hb48a81611cdf3cbdE
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.8.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_UnloadLibrary referenced in function _ZN4sdl25video43_$LT$impl$u20$sdl2..sdl..VideoSubsystem$GT$21vulkan_unload_library17h58b8eee9cc1a8219E
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.8.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_GetVkGetInstanceProcAddr referenced in function _ZN4sdl25video43_$LT$impl$u20$sdl2..sdl..VideoSubsystem$GT$32vulkan_get_proc_address_function17h58e1447184d60ec7E
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.7.rcgu.o) : error LNK2019: unresolved external symbol SDL_JoystickRumble referenced in function _ZN4sdl28joystick8Joystick10set_rumble17he4c3e1c8dc08bac7E
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.7.rcgu.o) : error LNK2019: unresolved external symbol SDL_GameControllerRumble referenced in function _ZN4sdl210controller14GameController10set_rumble17h3e69a2dcd5d35e6fE
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.4.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_GetInstanceExtensions referenced in function _ZN4sdl25video6Window26vulkan_instance_extensions17h4ce0b850b33c9155E
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.4.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_CreateSurface referenced in function _ZN4sdl25video6Window21vulkan_create_surface17hc627ff4976a3433fE
          libsdl2-72911fbe28401a0a.rlib(sdl2-72911fbe28401a0a.sdl2.b1fc5psl-cgu.4.rcgu.o) : error LNK2019: unresolved external symbol SDL_Vulkan_GetDrawableSize referenced in function _ZN4sdl25video6Window20vulkan_drawable_size17h02243f8c7fd2b6d6E
          D:\foo\target\debug\deps\gui.exe : fatal error LNK1120: 8 unresolved externals
```
Please make sdl an optional dependency for `embedded-graphics-simulator` :)

jamwaffles, 2020-03-28:
Makes sense to me. Any thoughts @rfuest? I've opened #298 as a starting point.

rfuest, 2020-03-28:
Using the simulator might not be the best option for @Boscop, as mentioned in https://github.com/jamwaffles/embedded-graphics/issues/299#issuecomment-605461098. But making SDL an optional dependency is a good idea anyway and would also help when using the simulator to generate PNGs in CI.

PR #298 (2020-03-30): Make SDL2 an optional simulator dependency
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check  [... cut]
closed 2020-03-30 by jamwaffles