embedded-graphics/embedded-graphics #295: image_buffer size differs from size passed to SimulatorDisplay

Boscop, 2020-03-28:
- Version of embedded-graphics in use (if applicable): [version here]
```
embedded-graphics = "0.6"
embedded-graphics-simulator = "0.2"
```
- Target: [simulator] / [real hardware]
ImageBuffer for use with imgui on glium

## Description of the problem
The `image_buffer` size differs from the size passed to SimulatorDisplay (`content_region_size`).
This is bad because the texture is much larger than I want and thus uses more mem, and I use many textures. (But it renders fine with imgui.)

Output:
```
content_region_size width: 800, height: 52
image_buffer width: 3199, height: 207
```

## Test case (if applicable)

```rust
pub fn draw_texture(
	gl_ctx: &impl Facade,
	content_region_size: [f32; 2],
	m_texture_id: &mut Option<TextureId>,
	textures: &mut Textures<Rc<Texture2d>>,
) {
	fn draw_into_buffer(
		(width, height): (u32, u32),
	) -> Result<ImageBuffer<Rgb<u8>, Box<[u8]>>, core::convert::Infallible> {
		let mut display: SimulatorDisplay<BinaryColor> = SimulatorDisplay::new(Size::new(width, height));
		Text::new("Hello World!", Point::new(5, 5))
			.into_styled(TextStyle::new(Font6x8, BinaryColor::On))
			.draw(&mut display)?;

		let output_settings = OutputSettingsBuilder::new().theme(BinaryColorTheme::OledBlue).build();
		Ok(display.to_image_buffer(&output_settings))
	}

	let (width, height) = (content_region_size[0] as u32, content_region_size[1] as u32);
	println!("content_region_size width: {}, height: {}", width, height);
	let image_buffer = draw_into_buffer((width, height)).unwrap();

	// image_buffer size differs from content_region_size
	let (width, height) = (image_buffer.width(), image_buffer.height());

	println!("image_buffer width: {}, height: {}", image_buffer.width(), image_buffer.height());
	let data = image_buffer.into_raw().into_vec();

	let raw = RawImage2d { data: Cow::Owned(data), width, height, format: ClientFormat::U8U8U8 };
	let gl_texture = Texture2d::new(gl_ctx, raw).unwrap();
	let gl_texture = Rc::new(gl_texture);

	if let Some(texture_id) = *m_texture_id {
		textures.replace(texture_id, gl_texture);
	} else {
		*m_texture_id = Some(textures.insert(gl_texture));
	}
}

```

[image]

jamwaffles, 2020-03-28:
I'm not quite sure what you're getting at here. For the two numbers you mentioned, is `(800, 52)` the pixel size of the text and `(3199, 207)` the size of the entire canvas? Are you somehow trying to just get the sub-buffer with only the text in it?

Boscop, 2020-03-28:
@jamwaffles No, what I mean is: I'm getting `content_region_size` from imgui, it's the available area, and I want to render a texture that is exactly that size, so I pass it to this function. I create the `SimulatorDisplay` with this size. BUT the ImageBuffer that it returns has the size `(3199, 207)`. Any idea why?

Maybe it's related to my other issue? https://github.com/jamwaffles/embedded-graphics/issues/299

rfuest, 2020-03-28:
Because the simulator is primarily used for simulating small displays like on a PC monitor, some scaling is automatically applied when selecting a `BinaryColorTheme`. If you add `.scale(1).pixel_spacing(0)` when calling the output settings builder the `ImageBuffer` should be the same size as the `SimualtorDispay`.

Boscop, 2020-03-28:
@rfuest Thanks, that solved it :)
I thought the default values would lead to equal size of the output buffer..

closed 2020-03-28 by Boscop