embedded-graphics/embedded-graphics #299: Scaling is wrong with custom DrawTarget

Boscop, 2020-03-28:
I'm trying to render into my custom `BufferDisplay`, but the scaling is very wrong.
The code below produces this:
[image]

`content_region_size` is (800, 52).

It seems to render as if zoomed in to the upper left third of the buffer.
The logical pixels are much bigger than the physical pixels in the buffer, which is apparent in the text rendering, especially with the subpixel aberrations.

Here is a zoomed-in screenshot where you can see the individual pixels.
The glyphs should be 32 pixels high..

[image]

.. but they are only 9 pixels high and 8 wide:

[image]

Also, the rect should have a red stroke and green fill, and it has neither.
And for some reason the image appears repeated on the x axis (3 times).

Any idea why that is?

```rust
pub fn draw_texture(
	gl_ctx: &impl Facade,
	content_region_size: [f32; 2],
	m_texture_id: &mut Option<TextureId>,
	textures: &mut Textures<Rc<Texture2d>>,
) {
	fn draw_into_buffer(
		(width, height): (u32, u32),
	) -> Result<Vec<u8>, core::convert::Infallible> {
		let mut display = BufferDisplay::new(width, height);
		
		let style = PrimitiveStyleBuilder::new()
			.stroke_color(Rgb888::RED)
			.stroke_width(3)
			.fill_color(Rgb888::GREEN)
			.build();
		Rectangle::new(Point::new(0, 0), Point::new((width - 1) as _, (height - 1) as _))
			.into_styled(style)
			.draw(&mut display)?;

		Text::new("Hello World!", Point::new(5, 5))
			.into_styled(TextStyle::new(Font24x32, Rgb888::new(255, 255, 0)))
			.draw(&mut display)?;

		Ok(display.buffer)
	}

	let (width, height) = (content_region_size[0] as u32, content_region_size[1] as u32);

	let data = draw_into_buffer(clip_with_userdata, (width, height)).unwrap();

	let raw = RawImage2d { data: Cow::Owned(data), width, height, format: ClientFormat::U8U8U8 };
	let gl_texture = Texture2d::new(gl_ctx, raw).unwrap();
	let gl_texture = Rc::new(gl_texture);

	if let Some(texture_id) = *m_texture_id {
		textures.replace(texture_id, gl_texture);
	} else {
		*m_texture_id = Some(textures.insert(gl_texture));
	}
}

struct BufferDisplay {
	width: u32,
	height: u32,
	buffer: Vec<u8>, // len = width * height * 3
}

impl BufferDisplay {
	fn new(width: u32, height: u32) -> Self {
		Self { width, height, buffer: vec![0u8; (width * height * 3) as _] }
	}
}

impl DrawTarget<Rgb888> for BufferDisplay {
	type Error = core::convert::Infallible;

	fn draw_pixel(&mut self, pixel: Pixel<Rgb888>) -> Result<(), Self::Error> {
		let Pixel(Point { x, y }, color) = pixel;
		if (0 .. self.width as _).contains(&x) && (0 .. self.height as _).contains(&y) {
			let index = (x + y * self.width as i32) as usize;
			let pixel = &mut self.buffer[index .. index + 3];
			pixel[0] = color.r();
			pixel[1] = color.g();
			pixel[2] = color.b();
		}
		Ok(())
	}

	fn size(&self) -> Size {
		Size::new(self.width, self.height)
	}
}
```

Maybe it's related to https://github.com/jamwaffles/embedded-graphics/issues/295

It can't be that my texture rendering code is wrongly scaling it, because then 1. the font sub-pixel rendering wouldn't be wrong and 2. the scaling is correct when I use the same texture rendering code with imgui but instead render into the texture via a `SimulatorDisplay`: 
[image]

That's with `Font6x8`, which is correctly scaled. Zoomed-in screenshot:
[image]

(But if I draw to a `SimulatorDisplay`, the resulting `ImageBuffer` is much larger than the `SimulatorDisplay` (https://github.com/jamwaffles/embedded-graphics/issues/295), which is why I decided to render to this custom `BufferDisplay`).

rfuest, 2020-03-28:
`let index = (x + y * self.width as i32) as usize;` is this correct? Should this be multiplied by 3, because you have 3 bytes per pixel?

rfuest, 2020-03-28:
But otherwise this would be the preferred way of using e-g with glium. Using the simulator display is inefficient because it would be converted many times before being displayed: `SimulatorDisplay` -> simulator `Framebuffer` -> `ImageBuffer` -> `RawImage2d` -> `Texture2d`.

Boscop, 2020-03-28:
@rfuest Ah right, I forgot to multiply by 3 there..
Yes, I figured it would be faster to use this `BufferDisplay`.
Btw, I think it would be useful to have it included in this crate, to make it more apparent that this crate is not only useful for "embedded" graphics, but also general purpose software rendering of simple shapes & text..

Boscop, 2020-03-28:
Btw, I thought there must be a general purpose (hardware-agnostic) software rendering crate that supports the things that this crate supports plus alpha blending and anti-aliased lines (and maybe filled polygons).
But I haven't found any. Is this the best crate, that comes closest?
I just need to render anti-aliased lines, rectangles with stroke (and semi-transparent fill) and strings. So this crate is almost perfect but I'm really missing the semi-transparent fill possibility for rects.
I saw the `andrew` crate which supports semi-transparent fill but it doesn't build on windows (https://github.com/trimental/andrew/issues/4), and hasn't been updated in a year.
(Apart from that, I only found some bresenham crates but only with aliasing.)

jamwaffles, 2020-03-28:
I'm not sure to be honest. e-g is targeted at embedded/no-std systems with the simulator almost as an afterthought so probably isn't ideal for "proper" graphics. For drawing 2D graphics I think I'd reach for SDL. I don't know of any crates specific to your requirements off the top of my head.

rfuest, 2020-03-28:
Did you look at cairo and the `cairo-rs` rust bindings?

Boscop, 2020-04-01:
I haven't tried cairo but I'm happy with this crate so far, it's close to perfect for my use case :)
(The feature I'm looking forward to the most is semi-transparent rectangle fill.)

closed 2020-04-01 by Boscop