embedded-graphics/embedded-graphics #304: Thin rects don't use thick stroke, rect width increase is not continuous

Boscop, 2020-04-03:
When you render a rect that is itself "thin", e.g. where x0 and x1 are equal (that is essentially a vertical line), then it will be drawn with stroke 1, not the given stroke!
Also even if `x1 = x0 + delta` it won't happen unless delta is at least 3 pixels.
And then, if you increase delta one by one, the drawn rect stays the same until delta becomes 7, only then it actually is rendered with that width.

I noticed this when drawing a rect below a line, the rect will completely be covered by the line when delta is 0, when I use 5 as the rect's stroke width:
I took screenshots of the green rect drawn under the red line with increasing delta from 0 to 7:
[image]
Zoomed in:
[image]

I re-rendered the scene without the red line to check how wide the green rect actually is at each delta:
| Delta  | Green rect's width (px) |
| ------------- | ------------- |
| 0 | 1 |
| 1 | 1 |
| 2 | 1 |
| 3 | 5 |
| 4 | 5 |
| 5 | 5 |
| 6 | 6 |
| 7 | 9 |

These are not smooth/continuous width increases, they are "steppy".
Even with a stroke width of 1, the rect with delta 1 should have a width of 2 px, with delta 2 it should have a width of 3 etc.
With a stroke with larger than 1, I wish the green rect would always be as wide as `stroke_width.max(delta+1)` (stroke should extend to both sides of the real rect's border by `stroke_width/2` pixels) so it would not be covered by the red line and thus invisible, because in this situation, the red line and green rect convey important semantic info and the red line has to be drawn last.

rfuest, 2020-04-04:
For filled shapes the stroke is currently drawn on the inside of the shape and not centered on the edges. This means rectangles with `width` or `height <= stroke_width * 2` will be completely filled with the border color. Other stroke alignments will probably be added in the future (see #182), but until then you can offset the rectangle manually to achieve the same result.

I don't quite understand the "steppy" increase problem. Could you provide a code snippet that demonstrates this?

jamwaffles, 2020-05-14:
> Other stroke alignments will probably be added in the future

Stroke alignment was added in #331 and will most likely be released in 0.7.

@Boscop If you still have the demo code around, can you try your example against the latest master and let us know if the "stepping" still occurs?

jamwaffles, 2020-10-23:
Closing due to inactivity. If this issue persists with e-g 0.7.0-alpha.1, please reopen with some example code to reproduce the original issue.

closed 2020-10-23 by jamwaffles