embedded-graphics/embedded-graphics #321: Rotations for primitives

jamwaffles, 2020-04-13:
We have `.translate()`. It should be possible to `.rotate()` a primitive. 

1. How should the center of rotation be defined? It could use the bounding box, or allow specifying an arbitrary point.
1. The easiest way to do this is to use `f32`s in a 2x2 matrix, but this requires sin()/cos() which is super slow on targets without FPUs. `micromath` or `libm` might help.

rfuest, 2020-11-17:
In addition to allowing rotation by an arbitrary angle we could also add rotation by 90° increments. Apart from possible performance advantages, this result of this rotation will always have the same type as the original primitive.

What return type would `Rectangle::rotate` for an arbitrary angle have? We can't return a `Polygon`, because it uses a borrowed slice.

nesk, 2026-08-15:
I had to implement this to rotate a triangle, here's my take if you wonder. Note that it works for me but it might be a bad implementation for others since:

1. I'm depending on core intrinsics
2. I'm using sin/cos (which is fine for me because I'm on an ESP32 S3 and it has an FPU)

```rs
#![no_std]
#![feature(core_float_math)] // See https://github.com/rust-lang/rust/issues/137578
#![feature(core_intrinsics)] // See https://doc.rust-lang.org/core/intrinsics/fn.cosf32.html and https://doc.rust-lang.org/core/intrinsics/fn.sinf32.html

use core::f32::consts::PI;
use core::f32::math;
use core::intrinsics::{cosf32, sinf32};
use embedded_graphics::prelude::*;
use embedded_graphics::primitives::Triangle;

pub trait Rotate {
    fn rotate_centered(&self, by_degrees: f32) -> Self;
}

impl Rotate for Triangle {
    fn rotate_centered(&self, by_degrees: f32) -> Self {
        let center = Point::new(
            mean(self.vertices.map(|p| p.x)),
            mean(self.vertices.map(|p| p.y)),
        );

        let radians = by_degrees * PI / 180f32;
        let cos = cosf32(radians);
        let sin = sinf32(radians);

        let vertices = self.vertices.map(|point| {
            let Point { x, y } = point;
            let centered_x = (x - center.x) as f32;
            let centered_y = (y - center.y) as f32;
            let rotated_x = math::round(centered_x * cos - centered_y * sin) as i32;
            let rotated_y = math::round(centered_x * sin + centered_y * cos) as i32;
            Point::new(rotated_x + center.x, rotated_y + center.y)
        });

        Triangle { vertices }
    }
}

fn mean(values: [i32; 3]) -> i32 {
    math::round((values.iter().sum::<i32>() as f32) / 3f32) as i32
}
```