embedded-graphics/embedded-graphics #326: RFC: Splitting the embedded-graphics crate

rfuest, 2020-04-15:
In https://github.com/jamwaffles/embedded-graphics/pull/324#issuecomment-614203140 I suggested that it might be a good idea to reduce the coupling between the driver interface and the rest of e-g. It would be nice if we could break parts of the API that aren't directly related to the driver interface without requiring to bump the e-g version of every driver.

With the current `DrawTarget` trait this isn't possible, because it depends on huge parts of e-g, like primitives and styles. But if we limit `DrawTarget` to only support the most common features we could reduce its dependencies a lot. Here is an example how this could be implemented:
- Move `Point` and `Size` in new `eg-geometry` crate
- Move color module in new `eg-color` crate
- Change `DrawTarget` and move to `eg-hal' crate:
  ```rust
  trait DrawTarget<C> {
     fn size(&self) -> Size;
     fn draw_pixel(&mut self, p: Point, c: Color);
     fn draw_iter(&mut self, i: Iter<Item = (Point, Color)>);
     /// Draws all pixels inside the rectangle defined by `top_left` and `size` with the colors in `pixels`
     fn draw_recangle_iter(&mut self, top_left: Point, size: Size, pixels: Iter<Item = Color>);
     /// Fills all pixels inside the rectangle defined by `top_left` and `size` with a single color
     fn fill_rectangle(&mut self, top_left: Point, size: Size, color: Color);
  }
  ```

Drivers would only depend on these new crates and not the rest of e-g. And these crates are less likely to require breaking changes than the rest of e-g, so that drivers could be used for multiple e-g versions without changes.

We would loose the support for primitive level acceleration, like the current `DrawTarget` trait provides. I'm not sure if that's a big problem, because it's far less common to have support for these operations. And relying on other hardware drawing routines, e.g. for lines, could lead to differences in the output of the internal drawing routines and the ones provided by the driver/display.

jamwaffles, 2020-04-19:
Hoo boy, let's just break things. Sorry for the cop-out reply but I don't want to go through the effort of both changing `DrawTarget` and communicating why it's changing due to a crate split. This feels like a nuke-from-orbit approach. Until the community gets tired of upgrading every 20 minutes, I want to keep this as a single crate. Let's keep `DrawTarget` as it is as well.

rfuest, 2020-04-19:
The `DrawTarget` tait is hard to implement correctly, even for a simple rectangle:
https://github.com/almindor/st7789/blob/15fca386e36668fa106e61a73a2eb88c5f8a905f/src/graphics.rs#L64-L85
https://github.com/yuri91/ili9341-rs/blob/7f678b7e9f0d8c1c3ee35691d9f2c1aab43abc87/src/graphics.rs#L44-L80
I think the ili9341 implementation might even be broken for rectangles with stroke and no fill.

And any changes to e-g, like the soon to be added stroke alignment, will break every driver. Which might not even be noticed by the driver author because the stroke alignment will simply be ignored.

Some recent additions, like the `Points` iterators and the suggested `fill_area` and `stroke_area` method, might make it easier to implement a correct driver. But I'm not sure if that's enough. If you want to keep the `DrawTarget` as is how can we make sure that driver implementations behave the same as the reference implementation in e-g?

jamwaffles, 2020-04-20:
> Sorry for the cop-out reply

Apologies for being short. Lockdown is starting to get to me...

> how can we make sure that driver implementations behave the same as the reference implementation in e-g?

Until we can come up with something better, IMO that's on the driver authors to ensure. I'm ok releasing 0.7.0 with the same `DrawTarget`, then fixing it later. I also think it's worth getting feedback from the Rust embedded community on what the pain points for the current `DrawTarget` are, then designing a better interface around those requirements instead of designing it in somewhat of a vacuum (although I don't have visibility on what feedback you've got). Until then, the easiest (if annoying approach for driver authors) is to leave `DrawTarget` as is.

If whatever we come up with is conducive to splitting the crate I'm certainly not opposed to doing that, but doing so now would limit what we can do with `DrawTarget`. It's better to keep our options open, no?

rfuest, 2020-04-20:
> Until we can come up with something better, IMO that's on the driver authors to ensure.

OK, but if something doesn't work as expected the issues will probably be reported for e-g and not the drivers, because the API user can't easily tell that the problem isn't caused by e-g directly.

> I'm ok releasing 0.7.0 with the same `DrawTarget`, then fixing it later.

Does this mean that you don't want to fix #263 and #181 in 0.7?

jamwaffles, 2020-04-20:
> OK, but if something doesn't work as expected the issues will probably be reported for e-g and not the drivers, because the API user can't easily tell that the problem isn't caused by e-g directly.

Let's see what happens. Feel free to throw those issues to me if they start coming in.

> Does this mean that you don't want to fix #263 and #181 in 0.7?

Oops, I forgot those existed tbh. Lots of open issues...

Yes, let's fix those if we can. I think by "with the same `DrawTarget`" I mean to keep the API as similar as possible., i.e. trying to stay away from the fundamental changes suggested in the OP. If we _have to_ breaking-change the API to support #263 or #181 let's see what that looks like at least.

rfuest, 2020-04-20:
As a compromise between the OP and keeping the API similar I would suggest this:
- Replace the broken `draw_image` with something like the `draw_rectangle_iter`  (needs a better name) suggested in the OP
  This method would be used for images and font rendering.
- Add a `fill_rectangle(rect: Rectangle, color: C)` method
  This can easily be implemented because it only fills the given rectangle and doesn't need to handle any special styles.
- `fill_rectangle` can be use in the default impl of `draw_rectangle` and `clear`.
  This way these methods are accelerated by simply overriding `fill_rectangle` but it would still be possible to override the individual method if this provides additional performance gains.
- Both `fill_rectangle` and `draw_rectangle_iter` can provide default impls, so this should be only a minor breaking change.
- The docs for driver authors could mention that only overriding `fill_rectangle` and `draw_rectangle_iter` is preferred and that extra care needs to be taken if any other method is replaced by a custom impl.

jamwaffles, 2020-04-20:
Thanks for detailing that out. It feels like a good compromise to me. Just a couple of clarifications:

> Replace the broken draw_image

What's broken about it? It might've been mentioned before but I missed it.

> fill_rectangle can be use in the default impl of draw_rectangle

Would `fill_rectangle` also be used for stroke-only rects? Is the idea that it could be used to draw four rects for the border of an unfilled rect, etc?

Should we rename this issue or close and open another one around changing `DrawTarget`?

rfuest, 2020-04-20:
> What's broken about it? It might've been mentioned before but I missed it.

One discussion about this was in https://github.com/jamwaffles/embedded-graphics/pull/258. And `draw_image` can currently only be used to draw images that are using the same color type as the display (see #328).

> Would `fill_rectangle` also be used for stroke-only rects? Is the idea that it could be used to draw four rects for the border of an unfilled rect, etc?

Yes, this would mean that this only needs to be implemented once in e-g and not in every driver.

For opaque rectangles with fill and stroke we would need to decide if we use multiple fill rectangle calls or a single `draw_rectangler_iter` call. Choosing one might not be the optimal solution for every driver but at least provide a better performance than per pixel updates.

> Should we rename this issue or close and open another one around changing `DrawTarget`?

I'll open a new one, because the original ideas isn't going to be implemented and the previous comments aren't all relevant anymore.

closed 2020-04-20 by rfuest