embedded-graphics/embedded-graphics #328: TinyTGA: Cannot load image exported from GIMP

dbrgn, 2020-04-19:
- Version of embedded-graphics in use (if applicable): 0.6.1
- Target: PineTime (ST7789)
- If using real hardware: https://crates.io/crates/st7789

## Description of the problem/feature request/other

To test TGA support, I took [this](http://2.bp.blogspot.com/_4kkT22kuvoA/TTnBGfY90lI/AAAAAAAADQQ/UvqB41_C1sk/s1600/elephants-child-2.jpg) image, opened in GIMP and exported it in TGA format.

[elephant.tga.gz](https://github.com/jamwaffles/embedded-graphics/files/4499446/elephant.tga.gz)

However, when writing the image to the screen, it looks noisy:

[image]

Ferris is on there to show that the hardware setup generally works. Rendering https://github.com/jamwaffles/embedded-graphics/blob/master/tinytga/tests/rust-rle-bw-topleft.tga works as well.

## Test case (if applicable)

```rust
let elephant_data = Tga::from_slice(include_bytes!("../elephant.tga")).unwrap();
let elephant: Image<Tga, Rgb565> = Image::new(&elephant_data, Point::new(20, 20));
elephant.draw(&mut lcd).unwrap();

let ferris_data: ImageRawLE<Rgb565> = ImageRawLE::new(include_bytes!("../ferris.raw"), 86, 64);
let ferris: Image<_, Rgb565> = Image::new(&ferris_data, Point::new(100, 80));
ferris.draw(&mut lcd).unwrap();
```

rfuest, 2020-04-19:
The image format specified in the second line of your example doesn't match the format stored in the TGA file. The TGA file uses 32bpp RGBA color, which you can decode by using:
```rust
let elephant: Image<Tga, Rgb888> = Image::new(&elephant_data, Point::new(20, 20));
```
This will ignore the alpha channel, but you could save some memory by removing it in GIMP.

Support for drawing images with different color formats isn't great in e-g 0.6 and you will need to use
```
elephant.into_iter().map(|Pixel(p, c)| Pixel(p, c.into()).draw(&mut lcd).unwrap();
```
to draw the image.

dbrgn, 2020-04-27:
I see, thanks for the explanation (and sorry for the late reply)! A quick note / example in the docs would probably be really useful to other people trying out tinytga without much knowledge about the format, like me :slightly_smiling_face:

closed 2020-04-27 by dbrgn