embedded-graphics/embedded-graphics #329: Allowing for different font implementations

jamwaffles, 2020-04-20:
Spurred by https://github.com/jamwaffles/embedded-graphics/pull/324#issuecomment-616098849, the current `Font` trait is poorly designed to handle fonts that aren't spritemaps. To this end, perhaps we can:

1. Rename `Font` to `SpritemapFont` (or better name!)
1. Repurpose the `Font` trait to be a general interface to `Text` to allow mapping from a character to a font glyph. Here's a starter:

    ```rust
    trait Font<C> {
        type Iter: Iterator<Item = Pixel<C>>;

        fn char_to_glyph(&self, c: char) -> Self::Iter;
    }
    ```

    I _think_ `char` is the right choice here. It's UTF-8 and can do things like this:

    ```rust
    fn main() {
        let c = '🙂';

        // >>> 🙂 f642
        println!("{} {:x}", c, c as u16);
    }
    ```

1. `impl Font for SpritemapFont`

Open to other suggestions of course.

Questions:

1. [ ]  How would `SubImage` (or whatever else) from #281 work in terms of lifetimes and references to pixel data with the trait given above? This could use some research.

rfuest, 2020-04-20:
> 2\. Repurpose the `Font` trait to be a general interface to `Text` to allow mapping from a character to a font glyph. Here's a starter:  ...

If `char_to_glyph` doesn't provide any information about the size of the glyph, how could `Text` render a string using this function?

IMO the font implementation should work on `&str` and not individual chars. An advanced font implementation might support ligatures which needs access to multiple chars or at least have some internal state during font rendering.

Some further questions:
- [ ] How can we support text metrics and positioning for different font renderers? This is also relevant for #225 and #270.
- [ ] An external font renderer might require additional style settings. How can these be supported in a generic way?

jamwaffles, 2020-04-20:
Something like this perhaps (with less bad names):

```rust
trait Font<C> {
    type Iter: Iterator<Item = Pixel<C>>;

    fn text_to_pixels(&self, text: &str) -> (Rectangle, Self::Iter);
}
```

Where the returned `Rectangle` is the bounding box of the text along with a pixel iterator to fill that bounding box.

This also makes me think that `(Rectangle, Self::Iter)` could become a generally reusable struct which holds a size and a pixel iterator. This might help with #263?

Finally, should we consolidate #225 and #270 into a single issue to discuss the `bounding_box` (#270) and `origin` (#225) behaviours?

> How can we support text metrics and positioning for different font renderers?

What do you mean by metrics? Font baseline and other properties, or simpler things like the pixel size of the rendered string? Shouldn't positioning relative to a point or inside a bounding box be handled by `Text`? My current vision is to let `Font` impls provide a rectangle filled with only text pixels. This can then be inserted into wrapping shapes/behaviour. Do you see it a different way?

> An external font renderer might require additional style settings. How can these be supported in a generic way?

Do you mean in cases like `TextStyle::new(font, BinaryColor::On)` where the font needs to be able to do more than use `BinaryColor::On`?

If a font needs extra settings for behaviour outside of styling, they can be set something like this:

```rust
// Create a kernable font with kerning turned off
let font = KernedFontBuilder::new().disable_kerning().build();

Text::new("This is not kerned", Point::new(15, 45))
    .into_styled(TextStyle::new(font, BinaryColor::On))
    .draw(&mut display)?;
```

Please correct me if you're thinking of something else.

jamwaffles, 2020-07-02:
Bump! It would be good to get the breaking changes of any new font groundwork into 0.7. As I see it, that currently means:

* Adding this issue to #380 
* Renaming `Font` to `SpritemapFont` or `BitmapFont` (the latter sounds better IMO)
* Refactoring the `Font` trait to become a lot more generic in how it handles text in -> pixels out

We should also figure out if there are breaking changes required for text alignment/positioning. I don't have any ideas on how to tackle this at the moment, but maybe we should open a new issue or carry on the discussion in #225 as this issue has gone off the rails a bit from my last rambling comment.

If you have grander plans for fonts as mentioned in https://github.com/jamwaffles/embedded-graphics/issues/225#issuecomment-568510001, I think we should probably close some existing font/text issues and open new one(s) to clean up some cruft.

rfuest, 2020-07-17:
After #397 I'm convinced that we should include at least some font changes in 0.7. I would like to work on this after `SubImage` is finished. I'll try to clean up the issues and answer the questions in your comment from April once I'm starting to work on this.

jamwaffles, 2020-11-16:
Is this closed by #470 or is there still more to do?

rfuest, 2020-11-16:
I did abandon the idea of using `SubImage` for the fonts at the moment. We could at some point reconsider using `SubImage` for fonts with > 1 BPP, but for that to be really useful we would need alpha blending. And if we add this in the future we might be able to implement it as a separate font renderer without breaking the existing one, now that we support multiple renderers.

closed 2020-11-16 by rfuest