embedded-graphics/embedded-graphics #330: DrawTarget improvements

rfuest, 2020-04-20:
As previously discussed in #326, the following changes to the `DrawTarget` trait were suggested:
- Replace `draw_image` by `draw_rectangle_iter(&mut self, rect: &Rectangle, iter: Iterator<Item = C>)`
  `draw_rectangle_iter` can be used to fill a rectangular region of the target with the colors provided by the `iter`. This can e.g. be used for images and font rendering.
-  Add a `fill_rectangle(&mut self, rect: &Rectangle, color: C)` method
  This fills the given rectangle with a single color, which might be more efficient that using `draw_rectangle_iter` if the display supports filling of memory regions.
- Change `draw_rectangle` to use `fill_rectangle` or `draw_rectangle_iter`
  For opaque styled rectangles without a stroke the default impl of `draw_rectangle` can call directly `fill_rectangle` with `styled.primitive` as the rectangle.
  Styled rectangles without a fill can be drawn using 4 individual `fill_rectangle` calls.
  Drawing rectangles with fill and stroke can be implemented using 2 overlapping `fill_rectangle` calls, 5 non overlapping `fill_rectangle` calls or a single `draw_rectangle_iter` call.
- Add docs for driver authors that suggest to implement `fill_rectangle` and `draw_rectangle_iter` instead of the other styled primitve `draw_...` methods.

Open questions:
- [ ] `draw_rectangle_iter` should be able to handle iterators that are shorter than `rect.size.width * rect.size.height` without leaving the display in an invalid state.
   Should we provide any method to make this easier for driver implementations or simply state this in the docs?
- [ ] `draw_rectangle_iter` needs a better name, that can't be easily confused with `draw_rectangle`
- [ ] Decide if `draw_rectangle` for styled rectangles with fill and stroke should be implemented using `fill_rectangle` or `draw_rectangle_iter`
  Using 2 overlapping `fill_rectangle` calls might not be a good idea if we plan to support transparency in the future.
- [ ] DrawTarget::size -> #263

jamwaffles, 2020-04-20:
Thanks for opening a new issue. Some comments:

> Decide if draw_rectangle for styled rectangles with fill and stroke should be implemented using fill_rectangle or draw_rectangle_iter
> Using 2 overlapping fill_rectangle calls might not be a good idea if we plan to support transparency in the future.

Overlapping rects would also be pretty inefficient, so I'd either go with the now-edited-out 5 rects approach, or the current `Rectangle` iterator over stroke/fill. Whatever we choose should be able to handle rounded corners when #83 is done.

> draw_rectangle_iter needs a better name, that can't be easily confused with draw_rectangle

`draw_contiguous_iter`, with `draw_bounded_iter` renamed to `draw_sparse_iter`? That's about all I've got at the moment.

> Add docs for driver authors that suggest to implement fill_rectangle and draw_rectangle_iter instead of the other styled primitve draw_... methods.

Would it be worth adding `#[deprecated]` notices to the other methods? That would provide a less painful migration path where 0.7 deprecates and 0.8 removes them.

> Should we provide any method to make this easier for driver implementations or simply state this in the docs?

Do you have an idea of what that method might look like? I'm leaning towards making driver authors aware of it instead of providing a method.

> Would something like draw_bounded_iter(&mut self, bound: Rectangle, iter: Iterator<Item=Pixel<C>>) be useful?

Oh, that's what I thought `draw_rectangle_iter` did. In any case, differentiating between contiguous and non-contiguous iterators might be a useful optimisation. Would it be dangerous to include it with a default impl to `draw_rectangle_iter`?

> DrawTarget::size -> #263

Did you get any further than the last comment you left on that thread? Is there some `Into<>` magic we can try?

rfuest, 2020-04-20:
> Overlapping rects would also be pretty inefficient, so I'd either go with the now-edited-out 5 rects approach, or the current `Rectangle` iterator over stroke/fill.

If communication if the bottleneck and the display provides a hardware accelerated fill command the overlapping rects could be the fastest. But I think one of the other approaches will be better for a larger number of cases.
What do you mean by "now-edited-out 5 rects approach"? It's still mentioned in the OP. Or do you mean something else?

> Whatever we choose should be able to handle rounded corners when #83 is done.

Didn't we discuss this and came to the conclusion that the corner radius is a parameter of the shape and not a style property? Which means that `RoundedRectangle` should be a separate struct.

> Would it be worth adding `#[deprecated]` notices to the other methods? That would provide a less painful migration path where 0.7 deprecates and 0.8 removes them.

I thought you wanted to keep these methods. Deprecating them now to remove them later doesn't make a lot of sense IMO. There are only a few drivers that use these features so this would only affect a very limited number of people. And if we decide to remove them we should just remove them now.

> Do you have an idea of what that method might look like? I'm leaning towards making driver authors aware of it instead of providing a method.

I'm not sure yet, but I would prefer this function not to panic if the iterator is to short. We will need to find a solution for the default impl anyway and maybe this could also be exported to be used by the drivers. An iterator adapter that simply repeats the last color could be a possible solution. But the most important aspect is to document the expected behavior when a too short or empty iterator is passed to this function.

> Oh, that's what I thought `draw_rectangle_iter` did. In any case, differentiating between contiguous and non-contiguous iterators might be a useful optimisation. Would it be dangerous to include it with a default impl to `draw_rectangle_iter`?

`draw_rectangle_iter` draws all pixels inside the `rect`, which `draw_bounded_iter` doesn't. So it's not possible to implement `draw_bounded_iter` with `draw_rectangle_iter`, but the other way around would work.

> Did you get any further than the last comment you left on that thread? Is there some `Into<>` magic we can try?

It's been a while since I looked at this. But I think the plan was to change `DrawTarget` to use an associated type for the color and only support one color format.

One advantage of this approach is that it's easier to support hardware color conversion (like in some STM32) this way. The display driver could use a typestate to represent the currently set up hardware color conversion and make sure that all drawing routines use this color format.

To support software color conversions we would need to provide additional methods, that might be implemented as extension traits on `Drawable`s or the `DrawTarget`, but I would need to look at the code before suggesting a good solution for this.

jamwaffles, 2020-04-20:
> If communication if the bottleneck and the display provides a hardware accelerated fill command the overlapping rects could be the fastest. But I think one of the other approaches will be better for a larger number of cases.

This is the default impl, so yeah more cases outweighs ultimate speed.

> What do you mean by "now-edited-out 5 rects approach"?

Hm, I saw it disappear. You must've moved it? But yes, I still mean the same thing in the OP :)

> Didn't we discuss this and came to the conclusion that the corner radius is a parameter of the shape and not a style property? Which means that RoundedRectangle should be a separate struct.

My sieve-like brain does not recall this lol. Can you find that discussion and add a note to #83? If not, I'll add a note there anyway.

> I thought you wanted to keep these methods. Deprecating them now to remove them later doesn't make a lot of sense IMO. There are only a few drivers that use these features so this would only affect a very limited number of people. And if we decide to remove them we should just remove them now.

To get a wider opinion on this, I was thinking of posting a message in the e-g Matrix room. Here's a draft:

```
For anybody that's written a display driver with e-g support, is it useful having methods to draw primitives like `draw_circle()`, `draw_triangle()`, etc? 

We're talking about changing `DrawTarget` to add some more general and hopefully more useful methods, but I'm stuck on whether to remove the primitive-specific methods or not and could use some input. 

There are a couple of impls out there for optimised versions `draw_rectangle` but they're pretty complex and are quite easy to get wrong, so it might be better to not have them in favour of letting e-g do more of the work internally for consistent results across drivers.

The proposed additions add methods for:

* Filling a given area with a solid colour
* Filling an area with contiguous colours (no transparency) from an iterator that provides pixel data
* (possibly) Filling an area with non-contiguous colours (on/off transparency) from an iterator that provides pixel data
```

Anything I should change?

> I'm not sure yet, but I would prefer this function not to panic if the iterator is to short.

Couldn't the method just return when the iterator is empty? Framebuffer drivers would leave remaining pixels alone, and immediate mode displays would simply not send the data. I don't see where a panic might come from.

> draw_rectangle_iter draws all pixels inside the rect, which draw_bounded_iter doesn't. 

Understood, thanks for explaining.

> It's been a while since I looked at this. But I think the plan was to change DrawTarget to use an associated type for the color and only support one color format.

So we fix `DrawTarget::size()` but leave the colour conversions for another PR, right?

rfuest, 2020-04-21:
> Anything I should change?

* The problems mentioned in the paragraph about `draw_rectangle` applies to all `draw_...` routines for styled primitives. Rectangles might be the easiest example to implement and getting pixel perfect circles and thick lines is much harder.
* It should also mention that future versions will add additional styling attributes that could be ignored by driver impls if they aren't carefully ported to the new version.
* I would use `rectangular area` instead of `area` in the proposed additions section

> Couldn't the method just return when the iterator is empty? Framebuffer drivers would leave remaining pixels alone, and immediate mode displays would simply not send the data. I don't see where a panic might come from.

Never mind, I wasn't thinking clearly when I wrote that. You are right that this is something the driver needs to handle and we don't have to care about.

> So we fix `DrawTarget::size()` but leave the colour conversions for another PR, right?

Let's first decide what to do with the drawing methods in `DrawTarget` and then make a plan on how to implement all changes to `DrawTarget`.

jamwaffles, 2020-04-21:
Thanks for the feedback. Here's a second go:

```
For anybody that's written a display driver with e-g support, is it useful having the methods
`draw_line`, `draw_triangle`, `draw_rectangle`, `draw_circle`, `draw_ellipse` and `draw_image` for
drawing specific e-g objects?

We're talking about changing `DrawTarget` to add some more general and hopefully more useful
methods, but I'm stuck on whether to remove the primitive-specific methods or not and could use some
input. 

There are a couple of impls out there for optimised primitive draw methods, however they're quite
complex, are easy to get wrong and are unlikely to be pixel-perfect with either other hardware impls
or e-g itself. Rectangles are reasonably simple, but pixel perfect lines, circles, etc are very
difficult to make consistent. 

There's also the concern that new style properties added to e-g in the future will be silently
ignored or improperly handled by the optimised driver impl, leading to further inconsistency between
displays, or even different primitive styles.

The proposed additions add methods for:

1. Filling a given rectangular area with a solid colour 
2. Filling an rectangular area with contiguous colours (no transparency) from an iterator that 
   provides pixel data
3. (possibly) Filling an area with non-contiguous colours (on/off transparency) from an iterator 
   that provides pixel data

This would then allow the removal of the methods listed in the first paragraph.
```

If that covers all the bases I'll post it in the e-g matrix room later.

> Let's first decide what to do with the drawing methods in DrawTarget and then make a plan on how to implement all changes to DrawTarget.

:+1:

rfuest, 2020-04-21:
> Thanks for the feedback. Here's a second go:

LGTM

jamwaffles, 2020-04-23:
Hopefully I've got an accurate account of what came out of the discussion on Matrix. It seems the consensus is that it's fine to remove `draw_line`, `draw_triangle`, `draw_rectangle`, `draw_circle`, `draw_ellipse` and `draw_image` and replace them with the proposed `DrawTarget` methods as mentioned above:

1. Filling a given rectangular area with a solid colour 
2. Filling an rectangular area with contiguous colours (no transparency) from an iterator that 
   provides pixel data
3. Filling an area with non-contiguous colours (on/off transparency) from an iterator 
   that provides pixel data

I haven't seen it mentioned either way, but another point that came up was the removal of `draw_pixel`. I think this might be a good idea. If drawing individual points is desired, perhaps e-g could provide a pixel struct that produces an iterator with one item in it. Another option is to pass in a `core::iter::once` and the appropriate `Size` (which is `Size::zero()` if I understand #182 correctly).

To distill the conversation so far, I believe the complete trait would look something similar to the following (with better names eventually):

```rust
trait DrawTarget<C> {
    fn size(&self) -> Size;

    /// Fill a given rectangle with pixels where not all pixels may be set (i.e. has binary transparency)
    ///
    /// E.g, `Circle`, `Triangle`, `Line`, `Text`.
    fn sparse_iter(&mut self, area: Rectangle, pixels: Iter<Item = Pixel<C>>);

    /// Fill a given rectangle with a contiguous list of pixel values (i.e. has no transparency)
    ///
    /// E.g, `Rectangle`, `Image`.
    fn fill_iter(&mut self, area: Rectangle, pixels: Iter<Item = Pixel<C>>);

    /// Fills all pixels inside the given rectangle with a single colour
    fn fill(&mut self, area: Rectangle, color: C);
}
```

Some other interesting points came out of the Matrix chat:

* We have the concept of a block, but what about lines and individual pixels? IMO these are special cases of a block so I don't think we need to specifically cater for them.
* [ ] Should `fill` be removed? The suggestion was that `core::iter::repeat` could be passed to `fill_iter` to produce the same end result. Another option is to provide a default impl that falls back to `fill_iter` but is overrideable. Taking the SSD1331 as an example, it can draw filled rectangles in hardware so having `fill` as an optimisation path would be useful for at least that display. I'm leaning towards keeping `fill` in the trait.
* [ ] Is there a way to efficiently handle single pixels, or do we even need to? I can't think of a display that can efficiently set individual pixels.

Thanks for bearing with me on this @rfuest :grimacing:. We got there in the end, and my ideas suck as usual :sweat_smile:

rfuest, 2020-04-23:
> I haven't seen it mentioned either way, but another point that came up was the removal of `draw_pixel`. I think this might be a good idea. If drawing individual points is desired, perhaps e-g could [...]

`Pixel` is a `Drawable` which could generate the appropriate iter for drawing single pixels. This will have a higher overhead than the current implementation, but should be OK because this isn't the preferred way of drawing pixels anyway. We could include an example that uses `iter::from_fn` to show an relatively easy way of drawing pixels without having to implement a iterator struct.

> We have the concept of a block, but what about lines and individual pixels? IMO these are special cases of a block so I don't think we need to specifically cater for them.

For individual pixels I would still keep the `draw_iter` method (see code below). For lines we should consider using `fill` to draw lines when they are axis aligned, which can be much faster than using `draw_iter`, which is necessary for thick slanted lines.

> * Should `fill` be removed? The suggestion was that `core::iter::repeat` could be passed to `fill_iter` to produce the same end result. Another option is to provide a default impl that falls back to [...]

Yes, we should definitively keep `fill`. The can even makes sense if there is no hardware fill support, like in the simulator, which overrides `draw_rectangle` with an optimized version.

My suggestion for `DrawTarget`: (don't pay attention to the names, I changed my mind half way through writing this)

```rust
trait DrawTarget {
    // using an associated type fixes #263 and color conversion shouldn't be handled here IMO
    type Color : PixelColor;

    fn size(&self) -> Size;

    // We could implement `sparse_iter` as:
    // (for images, circles, unfilled rectangles and text without bg)
    fn fill_from_sparse_iter(
        &mut self, area: &Rectangle, pixels: Iter<Item = Option<Self::Color>>
    );
    // This would guarantee the iteration order from top left to bottom right
    // and return `None` for transparent pixels. That should make it easier to
    // implement bundling of individual pixel updates into blocks.

    // But this wouldn't work for lines or other drawables that don't use the
    // top left to bottom right iteration order, so we also need the good old `draw_iter`:
    fn draw_iter(&mut self, pixels: Iter<Item = Pixel<Self::Color>>);
    // The drivers might still be able to optimize this by generating blocks,
    // but it's harder to implement because the next pixel could be anywhere.

    // If this only fills the given rectangle the item doesn't need to provide a coordinate IMO
    // and this way you could also use `target.fill_iter(&rect, iter::repeat(BinaryColor::On))`.
    // The driver can easily generate the required coordinates by using:
    // `area.points().zip(colors)` which I think is really neat.
    fn fill_from_iter(&mut self, area: &Rectangle, colors: Iter<Item = Self::Color>);

    fn fill_solid(&mut self, area: &Rectangle, color: Self::Color);
}
```

jamwaffles, 2020-04-23:
> Pixel is a Drawable which could generate the appropriate iter for drawing single pixels. 

> ... but should be OK because this isn't the preferred way of drawing pixels anyway.

Sounds good to me

> For lines we should consider using fill to draw lines when they are axis aligned

This is very true, but I was referring to _scanlines_. I don't think they're anything we need to specifically cater for at the moment though.

> Yes, we should definitively keep fill.

Great, we're on the same page :)

---

In your proposal, `DrawTarget` now takes three different impls of iter. How would that work for e-g items? Would they still only need to implement `Iterator<Item = Pixel<C>>`?

rfuest, 2020-04-24:
> This is very true, but I was referring to _scanlines_. I don't think they're anything we need to specifically cater for at the moment though.

I agree, a scanline is simply a 1px high rectangle.

> In your proposal, `DrawTarget` now takes three different impls of iter. How would that work for e-g items? Would they still only need to implement `Iterator<Item = Pixel<C>>`?

The main method items need to implement is `Drawable::draw` which is responsible for calling the correct methods in `DrawTarget`.

Some drawable won't technically need to implement iterators at all. A styled rectangle can be drawn by calling `fill_solid` multiple times in `draw`, without any custom iterator.

rfuest, 2020-04-30:
If you want to give it a try go ahead, I don't have much time to work on this at the moment.

rfuest, 2020-04-30:
If I find a spare moment I want to finish #314 first, before starting to work on something else. But if I should unexpectedly find many spare moments I'll let you know before starting to work on this.

jamwaffles, 2020-04-30:
Sounds good. Let me know if I can do anything to help.

jamwaffles, 2020-05-14:
FYI, I'm gonna grab this and have a go.

closed 2020-08-10 by jamwaffles