embedded-graphics/embedded-graphics #332: hello-world example in README.md is broken

rfuest, 2020-04-24:
The copy of the hello-world example in the readme is currently broken again.

Should we leave this issue open and fix this once before the release of the new stable version? Or should we reenable the readme check in master and comment the check out again before the release? Or is there a better permanent solution?

jamwaffles, 2020-04-24:
Let's just leave this issue open for now, but I'd certainly like to find a better permanent solution.

jamwaffles, 2020-06-11:
The `nalgebra_support` feature also has an outdated description:

```
- `nalgebra_support` - use the [Nalgebra](https://crates.io/crates/nalgebra) crate with `no_std`
  support to enable conversions from `nalgebra::Vector2` to `Coord` and `UnsignedCoord`.
```

PR #381 (2020-07-20): Cargo readme
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've ope [... cut]
closed 2020-07-20 by jamwaffles