embedded-graphics/embedded-graphics #338: Discuss making component min/max methods public

jamwaffles, 2020-05-09:
Off the back of https://github.com/jamwaffles/embedded-graphics/pull/336#discussion_r422515032, should we make the following methods `pub` (they are currently `pub(crate)`)?

- `Point::component_min`
- `Point::component_max`
- `Size::component_min`
- `Size::component_max`

rfuest, 2020-05-30:
Other possible candidates for addition:
* `Point::length_squared` (added in #353)
* `Point::component_mul`
* `Point::component_div`
* `Size::component_mul`
* `Size::component_div`
* `Point::swap_xy`
* `Size::swap_xy`

jamwaffles, 2020-05-31:
I thought `length_squared` and `swap_*` might make more sense as non-pub methods, but certainly can't hurt to expose them publicly to those who might need it.

Throwing my hat into the ring: Nalgebra also has methods like [`.x()`](https://docs.rs/nalgebra/0.21.0/nalgebra/base/type.VectorN.html#method.x) and `.y()` which return their respective components set to `1`, with all other components set to zero. I don't see a use for them internally but still might be useful.

rfuest, 2020-05-31:
> Throwing my hat into the ring: Nalgebra also has methods like [`.x()`](https://docs.rs/nalgebra/0.21.0/nalgebra/base/type.VectorN.html#method.x) and `.y()` which return their respective components [...]

Do I understand I correctly that you mean constructors like `Point::x()` and `Point::y()`? They could indeed by useful.

jamwaffles, 2020-05-31:
> Do I understand I correctly that you mean constructors like Point::x() and Point::y()?

Exactly that

rfuest, 2020-07-19:
Should we close this issue for now? I haven't added some of the suggested methods (`x`, `y` and `swap_xy`), but they didn't seem to be as useful as the others. If we decide to need them later we can open another issue.

jamwaffles, 2020-07-19:
Yeah, let's close it. It's pretty open ended so let's call it done for now.

closed 2020-07-19 by rfuest