embedded-graphics/embedded-graphics #339: Feature request: BinaryColor::Invert

bugadani, 2020-05-10:
My use case involves a binary display ( an ssd1306-based 128x64 display) and requires a sort-of "inverting" binary color. For example, drawing a line using this "color" would invert every pixel on that line. My main use case for this would be to draw text over unknown/changing backgrounds.

jamwaffles, 2020-05-10:
Hi, thanks for the feature request. Similar to #300 and #296, this will require some way of reading existing pixels on a display. We haven't given much thought to this yet AFAIK but the need is increasingly clear.

bugadani, 2020-06-17:
For anyone interested, this can be done without modifying embedded-graphics. (Although some modification of the underlying display driver may be necessary).

For an incomplete demonstration, see https://bugadani.github.io/embedded-graphics/ssd1306/performance/2020/06/17/embedded-graphics.html