embedded-graphics/embedded-graphics #355: Add TranslatedDrawTarget

jamwaffles, 2020-06-01:
See [this thread](https://github.com/jamwaffles/embedded-graphics/pull/342#discussion_r432995238).

Once #342 is merged, e-g needs a way to differentiate between a display's drawable area and a display's visible area. A suggestion in the linked thread was to add a `TranslatedDrawTarget` (name needs bikeshedding) to wrap another draw target in a "sub window" type object.

Note to self: read the thread again before commenting!

rfuest, 2020-06-01:
> ..add a `TranslatedDrawTarget` (name needs bikeshedding)...

The `TranslatedDrawTarget` was originally only intended to shift all drawing operations by a fixed offset. But I agree that it needs a new name, if it also changes the size of the drawing area. 

One thing to keep in mind when coming up with a name is that we might introduce additional wrappers around a draw target later, for example for mirroring or rotation. So the name for `TranslatedDrawTarget` shouldn't be to generic to avoid confusion with these other wrappers.

jamwaffles, 2020-06-01:
It would be wise to find a name in common with whatever we decide for #281. `SubImage` and `SubDrawTarget` are my current submissions.

rfuest, 2020-06-01:
> It would be wise to find a name in common with whatever we decide for #281. ``

Good point!

> `SubImage` and `SubDrawTarget` are my current submissions.

Sounds good, let's use these.

jamwaffles, 2020-06-02:
Reminder: revisit the doc comment text mentioned in https://github.com/jamwaffles/embedded-graphics/pull/342#discussion_r434092087 to clarify dimension behaviour.

bugadani, 2020-06-05:
So, is this `TranslatedDrawTarget` basically a clipping rectangle over a DrawTarget? If so, it's awesome, because I was about to open a feature request for this :)

jamwaffles, 2020-06-05:
As I understand it, yes. Something like this:

[image]

rfuest, 2020-06-05:
I didn't intend this to also apply clipping. There are possible use cases to allow drawing outside the `SubDrawTarget` area.

Implementing a clipping `DrawTarget` is a good idea and could work together with the `SubDrawTarget` to achieve the desired result. `ClippingDrawTarget` would only clip drawing operations outside the `Rectangle::new(Point::zero(), target.size()`. Nesting the two into a `ClippingDrawTarget<SubDrawTarget<T>>` would then allow to clip to any rectangle.

jamwaffles, 2020-07-26:
A couple of questions from https://github.com/jamwaffles/embedded-graphics/pull/409#issuecomment-663567621:

Can sub draw targets be larger than their parent? IMO this should be possible. If not, should we convey creation failure with a Result?
Can the subtarget be resized? If so, do we want to make a variant that can't which is just an offset of the parent DrawTarget? I'm not sure this really gives any benefit and just complicates the API so I'd say we just have one that can be resized for now, IMO.

And another one:

What would a true `SubDrawTarget` implementation look like? Using `TranslatedDrawTarget<ClippingDrawTarget>` wouldn't allow the sub target to be smaller than the original display. Would we add `SubDrawTarget` as a third target to be used like `SubDrawTarget<ClippingDrawTarget>`? If this is the case, wouldn't we need yet another target like `SubDrawTarget<AllowOverflowDrawTarget>` to remove the clipping behaviour?

rfuest, 2020-07-26:
> Can sub draw targets be larger than their parent? IMO this should be possible. If not, should we convey creation failure with a Result?

IMO there are arguments for and against allowing this. Can you give an example where it would be required to allow this?

> Can the subtarget be resized? If so, do we want to make a variant that can't which is just an offset of the parent DrawTarget? I'm not sure this really gives any benefit and just complicates the API so I'd say we just have one that can be resized for now, IMO.

What do you mean by "resized" in this context? Allowing to change the size at all or allowing to change the size after it was constructed?

> Using `TranslatedDrawTarget<ClippingDrawTarget>` wouldn't allow the sub target to be smaller than the original display

It would, because `ClippingDrawTarget` will have a `clip_area` or `size` field that defines the size of the draw target. But the correct type would be `ClippingDrawTarget<TranslatedDrawTarget<T>>`, because the clipping needs to be executed after the translation.

jamwaffles, 2020-07-26:
Sorry, I wasn't very clear when I originally typed that out.

> IMO there are arguments for and against allowing this. Can you give an example where it would be required to allow this?

I agree. I don't think the sub should be allowed to be larger. My question here is how do we prevent that in code. Returning a `Result` or just `panic`ing or...

> What do you mean by "resized" in this context? Allowing to change the size at all or allowing to change the size after it was constructed?

Going back to my "window manager" example, this would be useful for resizing a window. However, when I originally wrote that comment we had different ideas on how this API should work. With your approach of short-lived subs as explained in https://github.com/jamwaffles/embedded-graphics/pull/409#issuecomment-663567621, my original question is invalid now.

> It would

Ah ok, that makes sense now! Sorry to keep banging on about the same point, but how do we support non-clipping sub targets?

rfuest, 2020-07-26:
> I agree. I don't think the sub should be allowed to be larger. My question here is how do we prevent that in code. Returning a `Result` or just `panic`ing or...

`panic`ing would be bad. `Result` would be an option, but I would suggest to use the same logic that is used by cairo: The new clipping area is the intersection of the old clipping area and the requested clipping area. This makes sure the new area can never be larger.

> Sorry to keep banging on about the same point, but how do we support non-clipping sub targets?

I'm a bit slow today, but I think I finally understand what you mean: A sub target that translates and also sets the size, but doesn't clip. Is that correct?

bugadani, 2020-07-26:
> The new clipping area is the intersection of the old clipping area and the requested clipping area.

That can cause some unexpected behaviour I think. The principle of least surprise would make me go with a `Result`.

jamwaffles, 2020-07-26:
> The new clipping area is the intersection of the old clipping area and the requested clipping area.

Sgtm :+1: 

> I'm a bit slow today, but I think I finally understand what you mean: A sub target that translates and also sets the size, but doesn't clip. Is that correct?

No problem! Yep, that's exactly what I meant :)

rfuest, 2020-07-26:
> No problem! Yep, that's exactly what I meant :)

And now I also start to understand my comment from June... 

My suggestion would be to do the following: Change `TranslatedDrawTarget` in PR #409 to `SubDrawTarget` and add an `size` field to it. The current `translated` method can stay and will pass the same size as the base draw target to the `SubDrawTarget` constructor. To support the use case you mentioned an additional method can be added to `DrawTargetExt` that takes an `offset` and `size` or a `Rectangle`. What would be a good name for that method? `sub_draw_target`? Or are `sub_draw_target` and `translated` to different from each other?

To allow more flexible clipping `ClippedDrawTarget` could also be extended to allow to clip to arbitrary rectangles instead of only allowing `Rectangle::new(Point::zero(), self.size())`. Otherwise you would need to use `SubDrawTaregt<ClippedDrawTarget<SubDrawTarget<T>>>` to implement this.

rfuest, 2020-07-26:
> That can cause some unexpected behaviour I think. The principle of least surprise would make me go with a `Result`.

IMO it is the expected behavior, because other APIs use the same behavior.

But I agree that there might be instances where this isn't the intended behavior. We could add additional methods, like `clip_checked(...) -> Result<...>` if this becomes a problem.

bugadani, 2020-07-26:
I have one more question: are 0 sized subtargets allowed?

rfuest, 2020-07-26:
> I have one more question: are 0 sized subtargets allowed?

Yes. Is there a good reason not to allow them?

bugadani, 2020-07-26:
I'm not sure if they are useful, but I don't have much experience with canvas APIs. But in the No case the error handling would have been interesting :)

jamwaffles, 2020-07-26:
> My suggestion would be to do the following.

That's a good approach. Let's see what that looks like in code and go from there.

> ... that takes an offset and size or a Rectangle.

Using `Rectangle` here is more consistent with the rest of the e-g API so I think we should go with that.

> What would be a good name for that method?
> ...
> Or are sub_draw_target and translated to different from each other?

They are very different indeed. It would be nice to be able to use the `-ed` suffix for both, but the only name I can come up with for `sub_draw_target` is `clipped` which isn't great. We could also perhaps go with `translated_target`, `sub_target?

> To allow more flexible clipping

That's a very good idea. If you want to make an attempt in #409 I'd be interested to see what you come up with. We can always leave it as a standalone feature for the future though - adding it shouldn't cause any breaking changes, right?

jamwaffles, 2020-07-26:
> I have one more question: are 0 sized subtargets allowed?

We allow other zero sized stuff so I think it's ok to allow this as well.

rfuest, 2020-07-26:
> They are very different indeed. It would be nice to be able to use the `-ed` suffix for both, but the only name I can come up with for `sub_draw_target` is `clipped` which isn't great. We could also perhaps go with `translated_target`, `sub_target`?

`translated_target` would work, but is very long. Maybe `cropped` instead of `sub_target`?

jamwaffles, 2020-07-27:
Nice, `translated` and `cropped` fit well together I think so let's go with that.

rfuest, 2020-07-29:
Closed by PR #409.

closed 2020-07-29 by rfuest