embedded-graphics/embedded-graphics #356: Add something to MockDisplay to check overdraw/diffs

jamwaffles, 2020-06-02:
As discussed in https://github.com/jamwaffles/embedded-graphics/pull/353#discussion_r432920157

In writing #342 I've found a need for checking for overdraw, so it would be useful to make the function (or functionality) in that comment thread part of the e-g test harness.

Another feature that would be very useful is some kind of diff. As of #342, draw targets won't necessarily use the primitive's iterator to output pixels, so we need to add a check to ensure their outputs are equal. A method of diffing the mock display will help ensure we keep the outputs the same.

rfuest, 2020-06-02:
I've started doing some clean up on the tests in general, which also adds some additional features to `MockDisplay`.

I'm planning to change `MockDisplay` to default to panicing when overdraw or out of bounds drawing occurs. IMO using this as the default behavior and adding an option to explicitly allow this makes more sense, because this way the test author needs to think if this is desired.

> Another feature that would be very useful is some kind of diff.

How would that be different than comparing two `MockDisplay`s by using `assert_eq!`?

jamwaffles, 2020-06-02:
> I'm planning to change MockDisplay to default to panicing when overdraw or out of bounds drawing occurs.

That's a good idea, although it would panic for cases [like the example code here](https://github.com/jamwaffles/embedded-graphics/pull/342/files#diff-e3a0e264cea75b4366d762f9c04c9edeR181-R191) which isn't good for that example. It might be useful to use `MockDisplay` in doc examples to make the outputs clearer, so a _clean_ way of telling `MockDisplay` to not panic would help in those cases. 

> How would that be different than comparing two MockDisplays by using assert_eq!?

Ah true, my bad.

rfuest, 2020-06-02:
> That's a good idea, although it would panic for cases [like the example code here](https://github.com/jamwaffles/embedded-graphics/pull/342/files#diff-e3a0e264cea75b4366d762f9c04c9edeR181-R191) [...]

Ah, I didn't think about examples at all.

I would use a setter, like `display.set_allow_overdraw(true);` to allow overdraw. Is that clean enough?

jamwaffles, 2020-06-02:
Yeah that sounds good. If it's not clear to the reader, they can always search the docs for its description :)

PR #357 (2020-06-03): Add MockDisplay features and cleanup font tests
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, c [... cut]
closed 2020-06-03 by rfuest