embedded-graphics/embedded-graphics #359: Add Game Boy Advance display driver

mogenson, 2020-06-04:
Hi. I implemented the `DrawTarget` trait for the Game Boy Advance display background and sprites: https://github.com/mogenson/embedded-graphics-gba

Let me know if you'd be willing to include a link to this crate in the [Display drivers](https://github.com/jamwaffles/embedded-graphics#display-drivers) section of the README.

rfuest, 2020-06-04:
That's great, I didn't expect e-g to be used on a GBA. If you want to have your driver included in the drivers list please open a PR that adds the links to `embedded-graphics/README.md` and `embedded-graphics/src/lib.rs`

rfuest, 2020-06-04:
I did take a quick look at your code and noticed something that might be a problem once the ongoing changes to the `DrawTarget` trait are stabilized. See #330 and #342 if you are interested in what is changing for 0.7.

You have implemented `DrawTaret<Bgr555>` and `DrawTarget<PaletteColor>` for `Display`, which won't be possible anymore for the new `DrawTarget`, because it was changed to use an associated type for the `Color`: https://github.com/jamwaffles/embedded-graphics/blob/37015ce2f60d553a6ee259df8cafab683a309579/embedded-graphics/src/draw_target.rs#L282-L284

I'm not very familiar with the GBA hardware, but if the display is in one of the modes at a time I would suggest to change your code to use a typestate to manage the current display mode. This way you would also get rid of the match in `draw_pixel` and it wouldn't be possible to draw with the wrong color type anymore (e.g. using `PaletteColor` for `Mode3` or `Mode5`).

mogenson, 2020-06-04:
Do you anticipate releasing version 0.7 soon? If so, I'll wait, implement the new `DrawTarget` trait, and submit a PR afterwards.

You're correct, the GBA display can only be in one mode at a time. The `Display` enum is just used as a wrapper because Rust's orphan rule means I can't implement the `DrawTarget<Bgr555>` trait directly for the `Mode3` or `Mode5` structs. Checking the enum variant means I don't have to have multiple trait implementations that are almost identical. However, the lack of duplicated code may not be worth the runtime type checking. I'll look into creating three wrapper types for `Mode3`, `Mode4`, and `Mode5` for the new `DrawTarget` implementation.

rfuest, 2020-06-04:
> Do you anticipate releasing version 0.7 soon? If so, I'll wait, implement the new `DrawTarget` trait, and submit a PR afterwards.

No, the final version will still take some time. But we want to release an alpha version soon, so that driver authors can try the new `DrawTarget` trait.

> I'll look into creating three wrapper types for `Mode3`, `Mode4`, and `Mode5` for the new `DrawTarget` implementation.

Sounds good, I think this should make the code cleaner and I doubt that it will increase code duplication much. But we always appreciate feedback on the driver API, if you think that we could make the trait eaiser to implement.

I'll have to fire up an GBA emulator some time to give this a try, becasue it seems to be a fun to play with.

mogenson, 2020-07-09:
Hi again,

I've refactored my crate to use wrapper types for the GBA display modes and sprite tile objects. I also switched to using the new `DrawTarget` trait interface.

However, I'm running into some compilation issues using the `tinytga` crate from the head of this repo. If I follow the usage shown in the `tinytga` tests:

```
let tga = Tga::from_slice(include_bytes!("../assets/background.tga")).unwrap();
let image: Image<_, Bgr555> = Image::new(&tga, Point::zero());
image.draw(display)?;
```

The build fails with the following errors:

```
error[E0432]: unresolved import `embedded_graphics::Pixel`
   --> /home/mike/.cargo/git/checkouts/embedded-graphics-4ee43c3234f6b38b/b6a2f03/tinytga/src/lib.rs:394:9
    |
394 |         Pixel as EgPixel,
    |         ^^^^^^^^^^^^^^^^ no `Pixel` in the root

error[E0308]: mismatched types
   --> examples/draw.rs:143:46
    |
143 |     let image: Image<_, Bgr555> = Image::new(&img, Point::zero());
    |                                              ^^^^ expected struct `embedded_graphics::image::ImageRaw`, found struct `tinytga::Tga`
    |
    = note: expected reference `&embedded_graphics::image::ImageRaw<'_, embedded_graphics::pixelcolor::Bgr555, _>`
               found reference `&tinytga::Tga<'_>`
```

Building the crates contained in the `embedded-graphics` workspace and running the test from this repo works correctly, so I'm not sure what's different about my crate.

jamwaffles, 2020-07-10:
Hmm. Strange. Do you have a public repo I can take a look at? That's probably the easiest way to debug this issue.

If you can't post a repo link, you may also want to make sure that:

1. You have the `graphics` feature enabled on your `tinytga` dependency.
2. Both `embedded-graphics` _and_` tinytga` are being included from the repository, not crates.io

mogenson, 2020-07-10:
Checkout the current master branch of https://github.com/mogenson/embedded-graphics-gba and let me know if the dependencies are setup correctly in `Cargo.toml`.

You'll need a nightly version of Rust to build along with the [cargo make](https://github.com/sagiegurari/cargo-make) and [cargo xbuild](https://github.com/rust-osdev/cargo-xbuild) crates. Cloning my repo and running `cargo make build` should show you the errors.

Note, the `cargo-make` and `cargo-xbuild` crates are not included as dependencies in `Cargo.toml` because they require a `std` environment there's a bug where they overwrite the `no_std` feature of other dependent crates. Also, I've built the `embedded-graphics` and `tinytga` crates with my nightly toolchain outside of this GBA project, so I don't think it's a nightly vs stable issue.

jamwaffles, 2020-07-11:
I had a quick look and changing the dependencies to use a `patch` directive fixes the issue for me:

```toml
[dependencies]
gba = { git = "https://github.com/rust-console/gba"}
embedded-graphics = "*"

[dev-dependencies]
tinytga = { version = "*", features = ["graphics"] }

[patch.crates-io]
tinytga = { git = "https://github.com/jamwaffles/embedded-graphics.git" }
embedded-graphics = { git = "https://github.com/jamwaffles/embedded-graphics.git" }
```

Not entirely sure why this works when the original approach didn't, but it resolves the `Pixel as EgPixel` error for me locally. I'd feel a bit weird publishing this with git dependencies and the wildcards in the `version = "*"` directives, however it should unblock your work on the crate.

mogenson, 2020-07-11:
Thanks for finding a workaround for this weird `cargo` behavior!

I don't think crates.io allows git dependencies, so I'll wait for new stable releases of `embedded-graphics`/`tinytga`/`gba` to publish.

Feel free to close this issue and I'll submit a PR when `embedded-graphics` version 0.7 is out.

jamwaffles, 2020-07-11:
No problem! I'm glad it works for you as well.

> Feel free to close this issue and I'll submit a PR when embedded-graphics version 0.7 is out.

Sounds good! Thanks for using embedded-graphics :grin:

closed 2020-07-11 by jamwaffles