embedded-graphics/embedded-graphics #361: Split primitives into multiple files

jamwaffles, 2020-06-05:
I think it's about time we did this. I was doing some work towards #343 which would likely make the primitives files even longer, spurring me to create this issue.

I'm going to hopefully create a PR addressing this asap unless @rfuest is working on this already.

rfuest, 2020-06-05:
I haven't started working on this yet. The next thing I wanted to work on in #314 is the `triangle` module refactoring, which would overlap with this work. Refactoring the code twice in a row seems wasteful. Do you want to also clean up the code in the process or should I split the primitive when I get around to cleaning up the code?

jamwaffles, 2020-06-05:
Yeah doing it twice does seem wasteful. If you don't mind I'll do that as part of the work to close this issue.

rfuest, 2020-06-05:
I don't have time to work on this myself ATM, so it will probably be implemented earlier if you include this with the other work in this issue.