embedded-graphics/embedded-graphics #364: Triangles don't draw negative pixels

jamwaffles, 2020-06-05:
## Description of the problem/feature request/other

The `Triangle` points iterator contains a check for `point.x >= 0 && point.y >= 0` which is a holdover from pre-0.7 behaviour. Pixels with negative coordinates should still be iterated over even if they're not visible.

## Test case (if applicable)

```rust
#[test]
fn off_screen_still_draws_points() {
    let off_screen = Triangle::new(Point::new(10, 10), Point::new(20, 20), Point::new(30, -30));
    let on_screen = off_screen.translate(Point::new(0, 35));

    assert!(off_screen
        .points()
        .eq(on_screen.points().map(|p| p - Point::new(0, 35))));
}
```

rfuest, 2020-06-05:
Interesting, I thought we had fixed this before in #308 and #309.

jamwaffles, 2020-06-05:
Yes, I vaguely remember doing something about this a while back. I've folded a fix into #363 as if I don't, it'll probably conflict with all the file moving in that PR.