embedded-graphics/embedded-graphics #374: TGA image example code in documentation doesn't compile

Thra11, 2020-06-24:
- Version of embedded-graphics in use: 0.6.2
- Version of tinytga in use: 0.3.2
- Target: simulator

## Description of the problem

This example code doesn't seem to compile for me: https://docs.rs/embedded-graphics/0.6.2/embedded_graphics/#display-a-tga-image . I get the following error:
```
error[E0308]: mismatched types
  --> src/main.rs:79:34
   |
79 |             let img = Image::new(&tga, Point::zero());
   |                                  ^^^^ expected struct `embedded_graphics::image::image_raw::ImageRaw`, found struct `tinytga::Tga`
   |
   = note: expected reference `&embedded_graphics::image::image_raw::ImageRaw<'_, _, _>`
              found reference `&tinytga::Tga<'_>`
```

rfuest, 2020-06-24:
Did you enable the `embedded-graphics` feature for the `tinytga` crate?

Thra11, 2020-06-24:
Thanks for the quick response! I did not enable the `embedded-graphics` feature. I have now. I still get the error.

rfuest, 2020-06-24:
Sorry, I did make a mistake in my reply: The feature is called `graphics` and not `embedded-graphics`.

Thra11, 2020-06-24:
Thanks! That fixed it. I wonder if adding a sentence to the example mentioning the `graphics` feature would help. Something like "tinytga must have the `graphics` feature enabled".

rfuest, 2020-06-24:
> I wonder if adding a sentence to the example mentioning the `graphics` feature would help. Something like "tinytga must have the `graphics` feature enabled".

Yes, that's a good idea. I've opened an issue to keep track of this.

Thra11, 2020-06-25:
Thanks. I'll close this one.

PR #379 (2020-06-30): Clearer feature flags
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once y [... cut]
closed 2020-06-25 by Thra11