embedded-graphics/embedded-graphics #376: Is this crate a good choice for very small displays (9x9, 10x10)

kangalioo, 2020-06-24:
This is more of a question than a feature suggestion or bug report. I hope this is not too terrible of a place to ask.

I'm considering abandoning my custom-written display handling code for the [Novation Launchpads](https://www.google.com/search?q=novation+launchpad) in favor of this crate. Looking at some of the pictures, I saw mainly displays of medium sized displays, like around 100x100 and up. The devices I'm planning to use embedded-graphics on, are 9x9 and 10x10. Is embedded-graphics a good choice for such tiny displays, or am I better off writing my own code for it?

jamwaffles, 2020-06-25:
> I hope this is not too terrible of a place to ask.

Not at all, thanks for asking!

If you want to do more than just set pixels, I think e-g would be a pretty good fit for those displays. There's no lower bound on display dimensions :). You might struggle to make out smaller features (e.g. text) with fewer pixels but to me it makes sense to reuse the work done in this crate if you want to draw graphics.

kangalioo, 2020-07-08:
I added support for embedded-graphics to my MIDI controller interfacing library, which allows super cool stuff like this https://youtu.be/KlX4wxs1faE

I conclude: yes, embedded-graphics is a good choice even for very tiny displays

closed 2020-07-08 by kangalioo

jamwaffles, 2020-07-08:
I’m continually surprised by where embedded-graphics shows up. That’s awesome! Thanks for posting the vid 😄