embedded-graphics/embedded-graphics #377: Possibility of a new 0.6.x bugfix release

bugadani, 2020-06-27:
Latest stable currently calculates `Size` for bounding boxes incorrectly, and has possibly other mistakes. I would like to take some time and gather bugfixes that were made to the codebase since release, and backport those to a maintenance branch. @jamwaffles  would you be interested in making a new bugfix release from that, while 0.7 is in the works?

bugadani, 2020-06-27:
To be fair, I don't understand what happened to 0.6.2, there are three commits marked as the release commit, each containing different stuff :D I'm not exactly sure which point I want to branch from, 0.6.2 on master has the Size issue fixed, but unreleased 👀

jamwaffles, 2020-06-28:
As #378 was just closed, I'll close this issue as well.

closed 2020-06-28 by jamwaffles