embedded-graphics/embedded-graphics #380: Release 0.7.0-alpha.1

jamwaffles, 2020-06-28:
Issues/features left before cutting 0.7.0-alpha.1:

* [ ] #365

(@rfuest please edit the above list with ones you'd like to see. There are others I haven't added as well but I cba to go through the issues list right now)

rfuest, 2020-07-17:
Sorry, I forgot about this issue. I've added some issues I would like to include in 0.7.0-alpha.1.

jamwaffles, 2020-08-10:
I've added all the above issues/PRs to the `0.7.0-alpha.1` project. I'd like to get back to using projects to track releases as they're a bit more automated, so I'm going to close this issue to avoid duplication. For anything else that should go in 0.7.0-alpha.1, please assign the issue/PR to the project.

closed 2020-08-10 by jamwaffles