embedded-graphics/embedded-graphics #382: GitHub shows old version as latest release

rfuest, 2020-06-29:
GitHub still considers 0.2.0 to be the latest release of e-g. The new layout now shows this on the front page of the repository:
[image]
This could make a visitor think that no new versions were released since 2018.

jamwaffles, 2020-06-29:
That is _really_ annoying. This new Github interface is... mixed good/bad/horseshit.

I can draft a new release of e-g for 0.6.2 at that given commit which might fix things.

rfuest, 2020-06-29:
I think you could create a release for the tag at: https://github.com/jamwaffles/embedded-graphics/tags

jamwaffles, 2020-06-29:
Good tip, thanks. I've created a new release with a low-effort message to ask the user to look at `CHANGELOG.md`.

Looks like this now:

[image]

rfuest, 2020-06-29:
That's better.
But this is one more thing we need to remember when doing a release, maybe we should create a release checklist.

jamwaffles, 2020-06-29:
We should indeed. How would you suggest doing that? An issue template? A markdown document in the project root? Something in the readme?

rfuest, 2020-06-29:
I wouldn't clutter the readme with this, because it isn't relevant for the normal user. IMO the markdown document in the project root would be the easiest solution.

jamwaffles, 2020-06-29:
Sounds good to me. I've repurposed this issue to create a checklist in the root.

Items I can think of off the top of my head:

Pre-release:

* `cargo-release` must be used to release a crate in this repo.
* Is the changelog up to date, particularly with breaking changes?
* Is the version in `Cargo.toml` for the crate to be released the same as crates.io? (cargo-release automatically bumps to the next version)
* Double check which level (major, minor, patch) of release this is

Post-release

* Is the tag created by `cargo-release` pushed to Github?
* Is a Github release made for the latest version?

rfuest, 2020-06-29:
I haven't done any release on crates.io yet, so I cannot comment on the details, but I think it looks good.
Maybe the pre-release checks should also contain that you need to make sure the latest CI build was successful.

jamwaffles, 2020-06-29:
Yeah that definitely makes sense

rfuest, 2020-06-29:
Should we announce all new releases on the matrix channel? Then this should also be add to the post-release section.

jamwaffles, 2020-06-29:
Ooh, yes! This is specific to me, but it may also be worth adding a "make @jamwaffles tweet about it, tagging @ rust-embedded" as well. I think I have a few embedded Rust followers who would appreciate knowing about it.

jamwaffles, 2020-11-07:
The Github CLI has a [`release` command](https://cli.github.com/manual/gh_release_create) which we may want to use to craft a GH release from the changelog. Not sure how feasible this is but it may be worth investigating.

rfuest, 2021-02-06:
The GitHub releases weren't updated for e-g `0.7.0-alpha.3` and core `0.2.0` yet.

rfuest, 2021-02-06:
The `simulator`, `tinytga` and `tinybmp` don't have any releases yet. Should we add pre-releases for the alpha versions or wait until the stable version?

jamwaffles, 2021-02-15:
> The GitHub releases weren't updated for e-g 0.7.0-alpha.3 and core 0.2.0 yet.

Just updated these with the associated changelog items. Thanks for pointing that out.

> The simulator, tinytga and tinybmp don't have any releases yet. Should we add pre-releases for the alpha versions or wait until the stable version?

We should probably get some alphas out in case anybody wants to do some testing. Happy to get those released if you think it's a good idea.

commit 8a3a0a02 (2021-02-16): Add release checklist

Closes #382
PR #551 (2021-02-17): Add release checklist
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Closes #382
commit 15239a91 (2021-02-17): Add release checklist (#551)

* Add release checklist

Closes #382

* Apply suggestions from code review

* Add note about origin name

* Use rebase flag for cleanliness

* Look for TODO/FIXMEs

* Add a vague note about other crates

* Tweak releasing instructions for prereleases

* Add action to check dependent version

* Typo fixes
closed 2021-02-17 by jamwaffles