embedded-graphics/embedded-graphics #384: Triangles with thick strokes

bugadani, 2020-07-04:
- Version of embedded-graphics in use (if applicable): 0.6.2
- Target: simulator

## Description of the problem/feature request/other

Triangle ignores stroke thickness

## Test case (if applicable)

```rust
use embedded_graphics_simulator::{
    BinaryColorTheme, OutputSettingsBuilder, SimulatorDisplay, Window,
};

use embedded_graphics::{
    pixelcolor::BinaryColor, prelude::*, primitives::Triangle, style::PrimitiveStyle,
};

fn main() -> Result<(), core::convert::Infallible> {
    let mut display: SimulatorDisplay<BinaryColor> = SimulatorDisplay::new(Size::new(128, 64));

    let output_settings = OutputSettingsBuilder::new()
        .theme(BinaryColorTheme::OledBlue)
        .build();

    let thick_stroke = PrimitiveStyle::with_stroke(BinaryColor::On, 3);

    Triangle::new(Point::new(0, 0), Point::new(12, 0), Point::new(6, 12))
        .into_styled(thick_stroke)
        .draw(&mut display)
        .unwrap();

    Window::new("Triangle", &output_settings).show_static(&display);
    Ok(())
}
```

[image]

Haven't checked master very closely, only a cursory glance on the git history, but I haven't seen any explicit commit targeted towards triangles.

jamwaffles, 2020-07-04:
Thanks for reporting this. This is a known problem as we haven’t solved the issue of line joiners yet, resulting in both triangles and poly lines not supporting thickness at the moment.

We can probably repurpose this issue as a tracking issue for thick lines on triangles and poly lines.

bugadani, 2020-07-04:
Hm, I didn't know that, maybe worth a mention somewhere? But truth be told, I was just assuming it would work, haven't checked the docs beforehand.

PR #475 (2020-11-16): Triangles with thick stroke
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

Adds thick stroke support for `Triangle`s. 

Closes #384
closed 2020-11-16 by jamwaffles