embedded-graphics/embedded-graphics #394: Replace scripts with a task runner

rfuest, 2020-07-14:
e-g contains multiple scripts to generate files or to aid with CI. This collection of scripts isn't very well documented and the required parameters aren't obvious.

I would suggest to replace the scripts be using a task runner like [casey/just](https://github.com/casey/just/) or [sagiegurari/cargo-make](https://github.com/sagiegurari/cargo-make). This way all task can be defined and documented in one central file and we can also benefit from other features like dependencies between tasks.

jamwaffles, 2020-07-14:
> e-g contains multiple scripts to generate files or to aid with CI. This collection of scripts isn't very well documented and the required parameters aren't obvious.

I agree. Thanks for opening this issue.

I had a quick look at both of those linked projects and I'm leaning towards `just` over `cargo-make`. I like the idea of TOML command files, but the syntax is pretty verbose. `just` uses a custom file format, but doesn't look too complex as I think what we kind of want is "bash, but just a little bit more". Do you have a preference/reason not to use `just`?

rfuest, 2020-07-14:
No, I'm using `just` in another project and I really like it. It's focused on one task and doesn't try to be a solution for everything. The syntax is similar to a Makefile without some of the weirdness.

jamwaffles, 2020-07-15:
Alright, let's go with `just` then :). If you want to take this on please assign yourself to the issue so I don't step on your toes.

PR #410 (2020-07-24): Add justfile
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [ ] Check that you've added passing tests and documentation
- [ ] Add a simulator example(s) where applicable
- [ ] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [ ] Run `rustfmt` on the project
- [ ] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've ope [... cut]
closed 2020-07-24 by rfuest