embedded-graphics/embedded-graphics #399: Making expand and shrink methods for primitives public

rfuest, 2020-07-19:
As part of implementing the `fill_area` and `stroke_area` methods I would also like to make `expand` and `shrink` for the primitives public. 

But I'm not sure how to name the methods and if it's better to merge them into one. Instead of having two methods they could also be implemented as one that takes a signed value as a parameter (>0 expands, <0 shrinks). Would this make more sense or should we keep the current implementation? And if we decide to change it to a single method, how should this method be called? Vector graphics program often refer to the operation as offset, but that wouldn't be a good method name IMO, because it's not clear that the offset is applied to the outline and not the position. Maybe `offset_outline`?

jamwaffles, 2020-07-19:
Although I can't personally see a need for `expand` and `shrink`, the behaviour is already there so I don't see why they can't be public and help users who need the functionality.

`expand` and `shrink` aren't names I've seen used. We could perhaps go with `outset` and `inset` respectively, or go with a single method called `offset` that takes a signed value as you suggest. I don't have a strong preference either way which isn't useful for making a decision I know...

Naming aside, one major problem I can see is with polygons and polylines: outsetting them produces a different shape as some vertices are merged, and others have a bevel applied to them, adding another vertex to the output. Insetting creates a similar problem, but I think will only ever result in _fewer_ vertices. We could always just not impl offsetting for shapes that don't easily support it, but that might be surprising to end users. The only other crappy option I can think of is an `Offset` trait with an associated type specifying the output. Normal shapes just get `Self`, whilst polygons/polylines would probably need an iterator.

rfuest, 2020-07-19:
> Although I can't personally see a need for `expand` and `shrink`, the behaviour is already there so I don't see why they can't be public and help users who need the functionality.

One usage example would be drawing a focus outline around a widget in a UI library: `widget.bounding_box().expand(5).into_styled(outline_style).draw(&mut display)?;` (where the outline width is < 5 pixel, so this can't be drawn by using the outside stroke alignment)
In touch screen applications it could also be used to increase the size of a hit target to be larger than the drawn UI element to make it easier to hit.

> ...or go with a single method called `offset` that takes a signed value as you suggest.

I'm leaning towards this solution because offset seems to be the most common name. I thought that we used `offset` for position elsewhere, but it seems to be only used internally (`Image::offset`) and not in the API.

> Naming aside, one major problem I can see is with polygons and polylines: outsetting them produces a different shape as some vertices are merged, and others have a bevel applied to them, adding [...]

Yes, that's also a problem that I considered and colinear triangles are also a problem. As a first step I want to only implement this functions for the easy cases. These are useful for UI applications and we can deal with the other cases later.

jamwaffles, 2020-07-19:
> I'm leaning towards this solution because offset seems to be the most common name.

Sounds good to me, let's go with that.

> As a first step I want to only implement this functions for the easy cases.

Sounds good too. Is it worth making `offset` part of a trait?

rfuest, 2020-07-19:
> Is it worth making `offset` part of a trait?

Yes. Do you think `OffsetShape` is a good name?

jamwaffles, 2020-07-19:
Either that or `OffsetOutline`, take whichever one makes most sense to you :)

PR #400 (2020-07-19): Add OffsetOutline and StyledPrimitiveAreas
Hi! Thank you for helping out with Embedded Graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a simulator example(s) where applicable
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) and appropriate crate (`embedded-graphics`, `simulator`, `tinytga`, `tinybmp`) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `./build.sh` (Linux/macOS only) and make sure it passes. If you use Windows, check  [... cut]
closed 2020-07-19 by rfuest