embedded-graphics/embedded-graphics #405: Bounding boxes for Arc and Sector don't take the angles into account

rfuest, 2020-07-20:
The `Dimensions` trait impl for `Arc` and `Sector` returns the same bounding box as a circle with the same `top_left` and `diameter`. But the bounding box can be much smaller if `angle_sweep` is small.

jamwaffles, 2020-08-11:
This is trickier than I first thought so I've unassigned myself if someone else wants to take a look.

jamwaffles, 2020-08-17:
Note: https://github.com/jamwaffles/embedded-graphics/pull/431#discussion_r471510430

Remember to fix styled bounding box for arc/sector also.

cedeber, 2022-09-30:
Hello, any news about this one? I've seen you've started something on another branch (outdated now) but stopped at some point. I would like to see it fixed as I am currently doing crazy stuff with this crate: https://github.com/cedeber/framebuffer-worker. I could have a look if you'll please update the branch - so that I can fork - and give me some advices where to continue.

jamwaffles, 2022-09-30:
I'd be very happy if someone else wanted to take a look at fixing this. I did a quick skim of my PR and it's so out of date I think it's better to start from scratch, perhaps using that PR as a reference instead of a base to fork from.