embedded-graphics/embedded-graphics #412: Add additional examples for sub draw targets

rfuest, 2020-07-29:
The docs for `DrawTargetExt` currently only contain examples for the individual sub draw targets. These examples don't demonstrate how multiple sub draw targets can be combined. An example for `Clipped<Translated<_>>` should be added to the `DrawTargetExt` documentation.

Additionally a simulator example should be added, which demonstrates the individual sub draw targets, but also useful combinations of sub draw targets.

jamwaffles, 2020-07-29:
We may want to add that to `DrawTarget` instead, or at least mention and link to `DrawTargetExt`. The way the docs are rendered, the ext trait isn't as discoverable as `DrawTarget` itself.

rfuest, 2020-07-29:
The `draw_target` module docs would also be a good place to provide an overview of `DrawTarget`s and sub draw targets.

dobleuber, 2024-10-04:
Hey! I'd like to start working on this project. I am looking for a no_std project for learning and contributing.
Could I work on this example?

rfuest, 2024-10-04:
Sure, contributions are always welcome. The examples haven't been looked at in a long time and a pair of fresh eyes would help us spot things that are missing or could be improved. It's sometimes hard to judge what part of the docs and examples is lacking if you have been working on the project for years.