embedded-graphics/embedded-graphics #413: Remove possible panics from `tinybmp` and `tinytga`

rfuest, 2020-07-29:
`tinybmp` and `tinytga` use `panic` to handle some errors. This makes it possible to cause a `panic` in an application by generating an unsupported or malformed image file. If the application allows user supplied images this could be a problem and can't easily be prevented by the user of the crates.

Both `tinybmp` and `tinytga` can panic in the `IntoIterator` implementation if the `bpp` value is invalid. How should we handle this error? One option would be to use an enum for the `bpp` field and reject any unsupported values while the header is parsed, where an error can be returned. This would mean that we would reject these files even if the user doesn't want to use our iterator implementation. Is that a problem or are other value for `bpp` and the usage without our iter so uncommon that we can ignore this use case?

`tinytga` can also panic if there is an error while decoding the RLE compression. There is no easy way to report an error to the API user at that point, so I think the best (only?) option is to stop the iterator prematurely when an error occurs.

`ImageType::Empty` isn't supported by `IntoIterator` yet and will also cause a panic, but that is easily fixable by returning an empty iter.

rfuest, 2020-08-10:
@jamwaffles Can you comment on this? I would like to work on this issue and the other `tiny...` issues next.

jamwaffles, 2020-08-10:
> Can you comment on this? I would like to work on this issue and the other tiny... issues next.

Sorry for the delay. Will you be making breaking changes that should be in 0.7? I'd like to tie off at least an alpha pretty soon.

> Both tinybmp and tinytga can panic in the IntoIterator implementation if the bpp value is invalid. How should we handle this error?

The best I can come up with is an enum. I'm not sure it's worth supporting arbitrary numbers of bits per pixel, at least at the moment. The safety benefits of using an enum are worth the rigidity I think. Is there scope to use an enum for now, but add an `Other(u8)` variant in the future to cater for other bit depths? Not sure if it's even worth it but just putting the idea out there.

> so I think the best (only?) option is to stop the iterator prematurely when an error occurs.

There's [fallible-iterator](https://crates.io/crates/fallible-iterator) but I'd rather not add special dependencies. I think exiting early is the best/only solution, but it's a good enough failure mode IMO.

FWIW, according to <https://docs.fileformat.com/image/bmp/> `tinybmp` is missing 2 and 4 BPP support but that's another issue.

rfuest, 2020-08-10:
> Will you be making breaking changes that should be in 0.7?

Not for this issue. The remaining bigger breaking changes would be in the text support, like porting the internal font renderer to use sub images and adding support for external renderers.

> The best I can come up with is an enum.

That's what I had in mind.

> Is there scope to use an enum for now, but add an `Other(u8)` variant in the future to cater for other bit depths? Not sure if it's even worth it but just putting the idea out there.

Perhaps we could add a `non_exhaustive` attribute to the enum, just to be sure.

> I think exiting early is the best/only solution, but it's a good enough failure mode IMO.

Another option would be to repeat the last valid pixel, which would make sure that the entire image area is filled. But this wouldn't work if the error occurred on the first pixel, so perhaps exiting is the best solution.

> FWIW, according to https://docs.fileformat.com/image/bmp/ `tinybmp` is missing 2 and 4 BPP support but that's another issue.

I'll look into this. But don't think these formats are very common and probably mostly used for images with a 4 or 16 color palette, which aren't yet supported by `tinybmp`.

jamwaffles, 2020-08-10:
> The remaining bigger breaking changes would be in the text support, like porting the internal font renderer to use sub images and adding support for external renderers.

Is that #329? If so, please add it to the `alpha` project.

> Perhaps we could add a non_exhaustive attribute to the enum, just to be sure.

Forgot about that, yeah that's a good idea.

> Another option would be to repeat the last valid pixel, which would make sure that the entire image area is filled. But this wouldn't work if the error occurred on the first pixel, so perhaps exiting is the best solution.

Is it possible to repeat `(0, 0, 0)` (black) instead? That might help with unexpected behaviour - the image fills its full area without leaving a strangely shaped transparent area if it exits early.

> But don't think these formats are very common

Agreed, just thought it was worth mentioning for completeness' sake. Can always add them later.

rfuest, 2020-08-10:
> Is it possible to repeat (0, 0, 0) (black) instead?

I didn't think we could without an additional trait bound to e.g. `Default`. But if we pad the `TgaIterator` with zeros after an error occurred we should get black pixels for any color format. If we later add support for images with pallets the first pallet entry would be used instead. Sounds like a good solution to me.

closed 2020-10-22 by rfuest