embedded-graphics/embedded-graphics #423: Images don't work correctly between color formats?

DianaNites, 2020-08-09:
- Version of embedded-graphics in use (if applicable): 0.6.2
- Target: - Target: QEMU 5.0.0
- If using real hardware: Custom

## Description of the problem/feature request/other

Okay, so first some background:

I'm using this crate for graphics in UEFI, which only knows the pixel format at runtime,
so I've implemented `DrawTarget` generically over `Into<Bgr888>` and just use `Rgb888::from` if need be.(blt and pixel bitmask currently unsupported). So far so good, the circle and some text example colors are correct, despite using `Rgb565`.

Then I wanted to display images, and found that `Image<Tga, Rgb888>` results in the colors being wrong. It compiles, it should be converting through `Into<Bgr888>`, yet the colors are wrong?

rfuest, 2020-08-09:
I think this should work like you described it.

Just to make sure the problem is in the color conversion and not in the TGA decoding: Does https://github.com/jamwaffles/embedded-graphics/blob/master/simulator/examples/assets/rust-pride.tga show the same color problems?

DianaNites, 2020-08-09:
Okay so i've tried both tga and bmp, of rust-pride and my own test image

Rust-pride.tga is incorrect, red and blue reversed.
[image]

rust-pride.bmp Bgr888 is even weirder, with
[image]
and this for Rgb888 
[image]

The same bmp color issue does not occur with my own test bmp image with Bgr888 though, oddly enough? Does bmp support different color formats and does mine just so happen to already be bgr, and the rust-pride one isnt, or something?

[image]

rfuest, 2020-08-09:
`rust-pride.bmp` is a 16 BPP `Rgb565` image, which should explain the weird colors for the BMP file.

Did you implement `draw_image` in your `DrawTarget` impl or are you using the default implementation?

Could you try to manually convert the colors? `image.into_iter().map(|Pixel(p, c)| Pixel(p, Rgb888::from(c))).draw(&mut display)`

DianaNites, 2020-08-09:
My `DrawTarget` uses the default.

I'll try manually converting the colors

DianaNites, 2020-08-09:
Manually converting the colors with `image.into_iter().map(|Pixel(p, c)| Pixel(p, Bgr888::from(c))).draw(&mut display)` did not work?

DianaNites, 2020-08-09:
re: rust-pride.bmp, using `Image<Bmp, Rgb565>` does indeed fix the rendering issue, the colors are still wrong though.

rfuest, 2020-08-09:
Did I understand you initial post correctly that you `DrawTarget` currently stores `Bgr888` colors into the framebuffer by converting them by something like: `color.into_storage()`? And that the `Rgb888::from`, you mentioned, is only used if UEFI reports another pixel format?

There were some issues with implementing `DrawTarget` for generic color types in the past. To debug the problem I would suggest to try tp implement `DrawTarget` for `Bgr888` (or whatever type the framebuffer uses) instead of a generic version.

DianaNites, 2020-08-09:
Yes you understand correctly, thats what I'm doing. It's generic over `Into<Bgr888>`, which covers all the other color types.

Also, a question, the `C` in `Image`, hows it used? Because the rust-pride bmp rendered and colored wrong without Rgb565 for `C`, so is `C` used for interpreting the images pixels *and* for writing them out? Wouldn't that mean it's impossible to use images that you don't know the exact format of in advance?

rfuest, 2020-08-09:
> Also, a question, the `C` in `Image`, hows it used? I ask because `rust-pride.bmp` rendered and colored wrong without `Rgb565` for `C`, so does that mean `C` is used for interpreting the images pixels _and_ for writing them out?

Yes, that's correct. That's why I suggested to manually convert the color type during drawing, because `DrawTarget::draw_image` was a last image addition to embedded-graphics `0.6` and never supported color conversion. And by implementing `DrawTarget` generically over `C` this can lead to wrong inferred typed by the compiler.

> Wouldn't that mean it's impossible to use images that you don't know the exact format of in advance?

Yes, that's also correct. I also noticed this while I did some work on the image support recently while working on some features for the next version of embedded-graphics. This isn't a huge problem for most embedded applications which can bundle the correctly formatted images, but it would be a nice to have feature none the less.

DianaNites, 2020-08-09:
In that case I think the color conversions are actually working fine, so long as the `C` for `Image` is correct

I think the problems are then:

1. The docs aren't clear that the `C` for `Image` is for the image data, not the display.(Which I now notice `draw_image` is generic)

2. It's impossible to use images you don't statically know the format of in advance. You can't just load and display an image.

The second one in particular is annoying, I'd like to display user-supplied images without a lot of hassle on their part. For now I guess a workaround is require them to use a specific format, but.. rather not.

rfuest, 2020-08-10:
> In that case I think the color conversions are actually working fine, so long as the `C` for `Image` is correct?

They should. Do they work in your application?

> The docs aren't clear that the `C` for `Image` is for the image data, not the display.(And I now notice `draw_image` is generic)

The next version of e-g contains many changes in this part of the code and the updated docs should make it clearer where the color type needs to be specified.

`draw_image` was not generic over `C` in the 0.6 version of `DrawTarget` (it uses the same `C` as the `DrawTarget`), which made it unusable if the image color type and display color type didn't match. The 0.7 version of `DrawTarget` was rewritten and doesn't contain `draw_image` anymore.

> It's impossible to use images you don't statically know the format of in advance. You can't just load and display an image.

I will look into this when I do some other work on `tinytga` and `tinybmp`. But this will only be added for e-g 0.7, because the recent changes in e-g make future versions of these crates incompatible with e-g 0.6.

rfuest, 2020-08-10:
I've opened #424 to track the unknown color format issue.

DianaNites, 2020-08-10:
> They should. Do they work in your application?

Oh, huh, nevermind i guess they don't? I thought they did but it was late so mis-saw? `rust-pride.bmp`, which is a `Rgb565` bmp right? does not color correctly unless `Image<Bmp, Bgr565>`, matching the Display, is used?

rfuest, 2020-08-10:
Can you share your `DrawTarget` impl and the code you use to draw the image? It's hard to comment on this without seeing the code.

DianaNites, 2020-08-11:
[`DrawTarget`](https://github.com/DianaNites/uefi-graphics/blob/main/src/lib.rs)

[Drawing code](https://github.com/DianaNites/bootloader/blob/42ea09fbaf28d7995403ca5dcbbac37d59211d48/src/main.rs#L62)

<details>
<summary>Output from current code
</summary>

[image]

</details>

rfuest, 2020-08-11:
This conversion causes the dim display for the non generic `DrawTarget` impl:
https://github.com/DianaNites/bootloader/blob/42ea09fbaf28d7995403ca5dcbbac37d59211d48/src/main.rs#L135-L136
The values returned by `r()`, `g()` and `b()` are not scaled to 0-255 for color types with less than 8 bit per color channel. The commented out version should correctly scale the color channels.

DianaNites, 2020-08-11:
Oh yeah, that. I was trying to see if maybe it'd help get the colors correct, but the red and blue are still reversed.

rfuest, 2020-08-11:
Can you change the text to a color with `R != B` to make sure that works correctly?

rfuest, 2020-08-11:
Did you try to use an array for `write_value`, which is suggested in the docs for the `uefi` crate: https://docs.rs/uefi/0.5.0/uefi/proto/console/gop/struct.FrameBuffer.html#method.write_value?
Example: https://github.com/rust-osdev/uefi-rs/blob/ea0c557fc9d361a850b467575e3d512765a4939b/uefi-test-runner/src/proto/console/gop.rs#L60-L65

DianaNites, 2020-08-11:
[`DrawTarget`](https://github.com/DianaNites/uefi-graphics/blob/851c28775ff4f139fe2a9e2c7decac8f5ccb52b8/src/lib.rs#L35-L45)

[Drawing](https://github.com/DianaNites/bootloader/blob/4d1692dfbc7646670b1c3187e1cdae90212a74a2/src/main.rs#L69-L71)

Didn't even notice my text had R == B.

Did both of your suggestions and, uh,

<details>
<summary>Thats not right, both texts are red, but they should be blue?</summary>

[image]

</details>

rfuest, 2020-08-12:
The UEFI GOP specs contain the following description:
[image]

According to this description the bytes in RGB modes should be organized as R, G, B in memory regardless of the endianness of the machine. By using `into_ne_bytes` on an x86_64 machine you will get the reversed order.

DianaNites, 2020-08-12:
> By using into_ne_bytes on an x86_64 machine you will get the reversed order.

But [`to_ne_bytes`](https://doc.rust-lang.org/stable/src/core/num/mod.rs.html#4330-4339) does nothing except transmute to an array, through a union. It does not change the order of anything?

DianaNites, 2020-08-12:
> By using into_ne_bytes on an x86_64 machine you will get the reversed order.

But [`to_ne_bytes`](https://doc.rust-lang.org/stable/src/core/num/mod.rs.html#4330-4339) does nothing except directly transmute to an array.(through a union for const) It does not change the order of anything?

<details>
<summary>Changing it to `to_be_bytes`, which I guess would "un-reverse" it? is substantially worse. And `to_le_bytes` would of course be a noop on x86</summary>

[image]

</details>

rfuest, 2020-08-12:
> But [`to_ne_bytes`](https://doc.rust-lang.org/stable/src/core/num/mod.rs.html#4330-4339) does nothing except directly transmute to an array.(through a union for const) It does not change the order of anything?

`Rgb888::new(0x11, 0x22, 0x33).into_storage()` results in an `u32` with a value of `0x00112233`. When you convert that value into an array by using `to_ne_bytes` on an `x86_64` machine you will get: `[0x33, 0x22, 0x11, 0x00]`.

> Changing it to `to_be_bytes`, which I guess would "un-reverse" it? is substantially worse. And `to_le_bytes` would of course be a noop on x86

The resulting array in this case would be `[0x00, 0x11, 0x22, 0x33]`, so you will need to remove the first element and not the last element of the array when copying it to the framebuffer.

DianaNites, 2020-08-12:
<details>
<summary>That seems to have done the trick? Thanks so much for your help.

Maybe the color types should provide a way to get an array in the appropriate format so they don't depend on endianness like this? Seems weird that `Rgb888` isn't actually laid out as `Rgb888` and all.

With the problem with this figured out and #424 for dynamic images, I guess this can be closed?
</summary>

[image]

</details>

rfuest, 2020-08-12:
> Maybe the color types should provide a way to get an array in the appropriate format so they don't depend on endianness like this?

That's a good idea. I've implemented this in PR #429.

> Seems weird that `Rgb888` isn't actually laid out as `Rgb888` and all.

That depends on the endianness of the machine and I agree that it can get weird.

> With the problem with this figured out and #424 for dynamic images, I guess this can be closed?

I'm glad that it works now and if you don't have any additional questions feel free to close the issue.

closed 2020-08-12 by DianaNites