embedded-graphics/embedded-graphics #442: Can't use unwrap on draw result

wjakobczyk, 2020-10-02:
- Version of embedded-graphics in use (if applicable): 0.6.2
- Target: real hardware
- If using real hardware: ST7920

## Description of the problem/feature request/other

When I use unwrap() on result of draw call (for Text), the compiler complains:
```
error[E0599]: no method named `unwrap` found for enum `core::result::Result<(), <impl DrawTarget<Rgb565> as embedded_graphics::DrawTarget<Rgb565>>::Error>` in the current scope
  --> src/ui/framework/test.rs:19:10
   |
19 |         .unwrap();
   |          ^^^^^^ method not found in `core::result::Result<(), <impl DrawTarget<Rgb565> as embedded_graphics::DrawTarget<Rgb565>>::Error>`
   |
   = note: the method `unwrap` exists but the following trait bounds were not satisfied:
           `<impl DrawTarget<Rgb565> as embedded_graphics::DrawTarget<Rgb565>>::Error: Debug`
```

If I don't use unwrap(), then compiler warns about unused result.

## Test case (if applicable)

```rust
use embedded_graphics::{
    fonts::{Font6x8, Text},
    pixelcolor::Rgb565,
    prelude::*,
    style::{TextStyle, TextStyleBuilder},
};

fn render(drawing: &mut impl DrawTarget<Rgb565>) {
    // Create a new text style
    let style = TextStyleBuilder::new(Font6x8)
        .text_color(Rgb565::YELLOW)
        .background_color(Rgb565::BLUE)
        .build();

    // Create a text at position (20, 30) and draw it using the previously defined style
    Text::new("Hello Rust!", Point::new(20, 30))
        .into_styled(style)
        .draw(drawing)
        .unwrap();
}
```

rfuest, 2020-10-03:
`DrawTarget` doesn't require the associated `Error` type to implement `Debug`. You could add an additional trait bound to get the code to compile:
```rust
fn render<T>(drawing: &mut T)
where
    T: DrawTarget<Rgb565>,
    T::Error: core::fmt::Debug,
{ ... }
```
or use `Text::new()....draw(drawing).ok().unwrap()`, which removes the requirement of having a `Debug` impl, but discards any additional error information.

But if this code part of some library, which the path in the error message suggests, a better solution would be to pass the error to the caller of `render` so it can be handled elsewhere:
```rust
fn render<T: DrawTarget<Rgb565>>(drawing: &mut T) -> Result<(), T::Error> {
    // Create a new text style
    let style = TextStyleBuilder::new(Font6x8)
        .text_color(Rgb565::YELLOW)
        .background_color(Rgb565::BLUE)
        .build();

    // Create a text at position (20, 30) and draw it using the previously defined style
    Text::new("Hello Rust!", Point::new(20, 30))
        .into_styled(style)
        .draw(drawing)
}
```

wjakobczyk, 2020-10-03:
Thanks, these are viable solutions but they don't fit my case. I would like just to panic when there is an error, with minimal code :)
Why can't the DrawTarget::Error just implement Debug? Are there any downsides?

wjakobczyk, 2020-10-05:
Thanks for the hints, I'll use is_ok() and then panic if it's not, though it's two additional lines of code :)
Out of curiosity now, why not make Error to require Debug directly in its declaration in DrawTarget?

rfuest, 2020-10-05:
> Thanks for the hints, I'll use is_ok() and then panic if it's not, though it's two additional lines of code :)

Why don't you use `.ok().unwrap()`, which is only 5 characters longer?

> Out of curiosity now, why not make Error to require Debug directly in its declaration in DrawTarget?

We don't use `Debug` internally and by adding it as an additional requirement we would limit the types that can be used as `Error`. Traits in the standard library do the same, for example the `FromStr` trait: https://doc.rust-lang.org/src/core/str/mod.rs.html#76-106. If an application requires an additional trait bound it's normally easy to add that but it's more complicated to work around a missing trait impl if the error type is defined in another crate.

wjakobczyk, 2020-10-06:
> > Thanks for the hints, I'll use is_ok() and then panic if it's not, though it's two additional lines of code :)
> 
> Why don't you use `.ok().unwrap()`, which is only 5 characters longer?

ok() consumes an error and I want to panic on it.

rfuest, 2020-10-06:
`ok()` converts an error to `Option::None` and calling `unwrap` on that will panic.

wjakobczyk, 2020-10-06:
Ah, right! You live and learn :)
Thanks!

closed 2020-10-06 by wjakobczyk