embedded-graphics/embedded-graphics #448: Implement From<Size> for Rectangle

rfuest, 2020-10-25:
Should we implement `From<Size> for Rectangle`? This would be equivalent to calling `Rectangle::new(Point::zero(), size)`, which seems to be a common operation (used 40 times in e-g according to a quick grep).

Or do you think the original version is easier to understand and the additional code verbosity doesn't matter?

rfuest, 2020-10-25:
Another alternative would be to add `Size::to_rectangle`.

jamwaffles, 2020-10-25:
`Size::to_rectangle` would be clearer than the magic `Size::new(...).into()`. It would reduce verbosity a bit, but beyond that I'm not sure I see much utility in this method although I have no objections to adding it in. Do you have a specific use case in mind? If you want to add it, IMO `Size::to_rectangle` should be `pub(crate)` at most.

rfuest, 2020-10-25:
> Do you have a specific use case in mind?

No, I just noticed that I've written the same code multiple times. One example where it looks cleaner IMO is this:
```rust
/* before: */ for point in Rectangle::new(Point::zero(), self.size()).points() { ... }
/* after: */  for point in self.size().to_rectangle().points() { ... }
```

>  If you want to add it, IMO `Size::to_rectangle` should be `pub(crate)` at most.

I've just use this in `tinytga`, so I would like to make it `pub` if we decide to add it. But I'll let you decide if we should add it at all. You can close the issue if you don't think it has enough benefits to be added to the public API.

jamwaffles, 2020-10-25:
> One example where it looks cleaner IMO is this:

That's a lot cleaner indeed.

I feel like I'm bikeshedding a bit here so let's add it in. It's a simple method with easily explained behaviour so not much of a burden on us to maintain it.

PR #577 (2021-04-06): Use bounding_box instead of Rectangle::new
This PR changes some `Rectangle::new(Point::zero(), ...)` expressions to use `bounding_box` instead, which makes the intent of these statements clearer.

These changes are a replacement for #448. While not all instances of `Rectangle::new(Point::zero(), ...)` can be replaced by `bounding_box` some of the other code is clearer without using `Size::to_rectangle`.

Closes #448
closed 2021-04-06 by rfuest