embedded-graphics/embedded-graphics #455: Use baseline for text position

rfuest, 2020-11-02:
The `position` field in `Text` objects currently specifies the top left corner of the texts bounding box. This can cause problems with other text renderers, where each glyph can have a different size and where the bounding box height depends on the actual text. For this reason other APIs use the left most point on the baseline to define the text position. This also makes it easier to mix different fonts, font sizes and even renderers on a single line of text.

To implement this change for the internal font renderer I would like to add a `BASELINE` constant to the `Font` trait, which defines the offset between the baseline and the top of glyph bounding box.

@bugadani: Will this be a problem for embedded-text? Do you need any additional information from the font renderer to correctly layout the text, other than the bounding box?

On another note: The BDF renderer based on the BDF parser from the `embedded-fonts` crate already works and only needs  more polish and better integration into e-g:
[image]
The WIP code can be found in my `embedded-fonts` fork: https://github.com/rfuest/embedded-fonts/tree/e-g

@jamwaffles: I did develop the code inside the `embedded-fonts` repo, because I needed to tweak the parser during development. Should I open a PR for `embedded-fonts` when I'm finished or should I put it somewhere else? IMO it would be good to make this an "official" e-g project, because it will also be a reference impl for the external font rendering API. Maybe it's time for a e-g organization?

bugadani, 2020-11-02:
Nice work! I think the bounding box and the baseline position should be plenty for what's currently in embedded-text. Actually, e-t doesn't just lay out the text but uses a slightly modified font renderer at the moment, but that's not an issue here.

jamwaffles, 2020-11-02:
> To implement this change for the internal font renderer I would like to add a BASELINE constant to the Font trait, which defines the offset between the baseline and the top of glyph bounding box.

SGTM. Do you want to keep the top left definition and allow the baseline to be specified as the start point, or replace it with the baseline entirely? I find the top left property a very easy way to position text but I agree it's quite brittle.

> On another note: The BDF renderer based on the BDF parser from the embedded-fonts crate already works and only needs more polish and better integration into e-g:

Wow very nice! I didn't realise it was that useable!

>  I did develop the code inside the embedded-fonts repo, because I needed to tweak the parser during development. Should I open a PR for embedded-fonts when I'm finished or should I put it somewhere else? 

That would be great if you're ok with the code being in that repo. I think it might be worth renaming that repo to `embedded-bdf` or `embedded-bdf-fonts` as it's focussed on BDFs only so far. What do you think?

> Maybe it's time for a e-g organization?

I think it is, yes. Here's my rough plan:

* @jamwaffles creates the `embedded-graphics-rs` organisation (or do we want to elide the `-rs` suffix? I don't think so IMO)
* @jamwaffles moves `embeddeds-graphics` and `embedded-fonts` (pending rename) into the organisation
* @jamwaffles prepares a PR in `embedded-graphics` to replace all links that point to jamwaffles/embedded-graphics to embedded-graphics-rs/embedded-graphics
* @rfuest or @jamwaffles does the same to `embedded-fonts` although it sounds like @rfuest has better ownership of that repo so maybe better for them to do it.

Later down the line we can split tinybmp, tinytga and embedded-graphics-simulator out of e-g into their own repos in the e-g-rs org, but let's not complicate things further now.

Should we get `embedded-text` into this organisation too? @bugadani would of course retain full ownership.

If we miss any links, I think github is good enough to redirect to the moved repos, at least in the short term.

Anything I've missed? There are permissions to work out but I think it might better to see where we stand once the migration is done and move on from there. I don't want to be a BDFL of this project by any means.

bugadani, 2020-11-02:
> Should we get embedded-text into this organisation too? @bugadani would of course retain full ownership.

That could be fun, if you think e-t is good enough for that :)

rfuest, 2020-11-02:
> Do you want to keep the top left definition and allow the baseline to be specified as the start point, or replace it with the baseline entirely?

As a first step I want to replace it entirely to make placement consistent across different font renderers. But I agree that the current behavior can be easier to use and I'll look into how this feature can be reimplemented later.

> That would be great if you're ok with the code being in that repo. I think it might be worth renaming that repo to `embedded-bdf` or `embedded-bdf-fonts` as it's focussed on BDFs only so far. What do you think?

I agree and IMO `embedded-bdf` would be a good name.

> @jamwaffles creates the `embedded-graphics-rs` organisation (or do we want to elide the `-rs` suffix? I don't think so IMO)

I'm on the fence about the name. While the `-rs` prefix is used often, I've mostly seen it used for projects with relatively short names without additional hyphens and `embedded-graphics` is already very long on its own. But if you think we need to emphasize that e-g is a rust project I'm also OK with the prefix. Just pick the one you like best.
 
> Should we get `embedded-text` into this organisation too?

Sounds good to me. It would be nice to also add support for BDF fonts in e-t in the future.

> Anything I've missed?

No, I think you plan is good. I'll handle the changes in `embedded-fonts`.

jamwaffles, 2020-11-02:
Something I missed in my comment above is the repository names. My suggestions:

* `<org>/embedded-graphics`
* `<org>/embedded-bdf`
* `<org>/embedded-text`
* `<org>/tinybmp`
* `<org>/tinytga`
* `<org>/simulator` (published as `embedded-graphics-simulator`)

Unless otherwise stated, the repositories have the same name as the crate. Should we call the sim repo `simulator` or `embedded-graphics-simulator`? I think the latter is quite redundant if it's under an `embedded-graphics` org.

rfuest, 2020-11-02:
> Unless otherwise stated, the repositories have the same name as the crate. Should we call the sim repo `simulator` or `embedded-graphics-simulator`? I think the latter is quite redundant if it's under an `embedded-graphics` org.

OK, `simulator` should be clear enough.

We also need a name for the docker repo we talked about, should we simply use `<org>/docker` or something else?

jamwaffles, 2020-11-02:
Either `docker` or `build-tools`? We might want to put other things in there in the future that aren't Docker-related, although no concrete examples of what those things might be come to mind.

jamwaffles, 2020-11-02:
After this amazing poll

[image]

I'm going to go with `embedded-graphics` as the org name. I'll get the ball rolling on this tomorrow when I'm a bit less tired.

rfuest, 2020-11-03:
Because this issue has been taken over by the org discussion I've renamed it and will create a new one for the original topic.

jamwaffles, 2020-11-03:
Sorry, this got pretty derailed didn't it...

rfuest, 2020-11-03:
> Sorry, this got pretty derailed didn't it...

It did, but my original comment provoked this.

rfuest, 2020-11-03:
> Either `docker` or `build-tools`?

Using the more generic `build-tools` sounds like a good idea to me. While I also can't imagine another use at the moment we will gain some flexibility for the future.

Should we also create an `examples` repo? The examples are currently in the `simulator` repo, but don't really belong there IMO.

jamwaffles, 2020-11-03:
Yeah, `build-tools` and `examples` are a good idea. I'll make those as empty repos for now and we can migrate stuff into them when we've got time

bugadani, 2020-11-03:
I believe the examples belong in the repository they are associated with. Arbitrarily extracting e-g's simulator based examples wouldn't serve much purpose except making it hard to keep the examples in sync with e-g itself.

I think the examples in e-g should be rewritten on top of (or replaced with the examples of) simulator, and maybe we should create some simple demo projects for ubiquitous hardware as separate repos. This would make code in e-g completely hardware independent, but the repo itself could still contain shiny sugarcoated demos that everyone could try on their PCs.

rfuest, 2020-11-04:
@bugadani I'm not quite sure what you mean. Which examples should be rewritten? The new `examples` repository will contain the examples that were previously included in the `simulator/examples` directory and are all simulator based.

Adding demo project for common hardware like the blue pill is definitely a good idea and has come up before. I think it would also be a good idea to add a step by step tutorial of how to get started with e-g on such a hardware platform, maybe in the form of a small e-g book.

bugadani, 2020-11-04:
@rfuest should we move the examples out of embedded-text as well to the same repository? They are all simulator-based. I would argue no, but doing so would be consistent if the e-g examples were moved there.

To be fair, I was wrong and mixed up the examples with the ones from the ssd1306 device driver - those are actually hardware dependent and I'm fine with that. Looking at it some more, I realized that the e-g crate contains no examples. I believe moving the ones from the simulator crate to e-g would make more sense (to me at least) than extracting the examples from *some* of the repositories. No rewrite necessary, that's my mixup :)

rfuest, 2020-11-04:
The e-g crate doesn't contain examples because it doesn't include a draw target that can be used to show the output to the user. We would need to add a dev-dependency to the simulator crate to add examples to the e-g crate, which would cause a cyclic dependency between the crates which has recently caused problems. By moving the examples to a separate repo/crate we can depend on all necessary crates in the e-g ecosystem without creating any cyclic dependency problems.

> @rfuest should we move the examples out of embedded-text as well to the same repository?

Other crates like tinytga and tinybmp also have their own examples, so I wouldn't remove any of the existing examples from e-t. But now that e-t is an official part of the e-g ecosystem I think that it would be a good idea to also add an e-t example to the new examples repo so that users are made aware of the features it provides.

bugadani, 2020-11-04:
Hmm I never considered cyclic dependencies. The split makes a lot more sense now, thanks. Although, if the core would be extracted from e-g, this problem would probably go away, wouldn't it?

We can probably make something flashy that showcases the interaction between the different crates. e-t also needs some love wrt. the examples, they are rather haphazardly thrown together. But that's something I have to face :)

rfuest, 2020-11-04:
> Hmm I never considered cyclic dependencies. The split makes a lot more sense now, thanks. Although, if the core would be extracted from e-g, this problem would probably go away, wouldn't it?

Interesting, I hadn't considered that the simulator is just another draw target that only needs to depend on the core crate. But do you think there would be a huge benefit of adding e-g only examples to the main e-g repo? IMO the external examples repo is the best place to put examples that demonstrate all features, regardless if they are provided by the main e-g crate or another crate in the ecosystem. And it has the additional benefit that we don't accidentally create any new weird (dev-)dependencies between the crates. 

> We can probably make something flashy that showcases the interaction between the different crates. e-t also needs some love wrt. the examples, they are rather haphazardly thrown together. But that's something I have to face :)

I just tried some e-t example and they are already look impressive. But all examples could use some improvements and now that we have more color constants it might be easier to make them look nicer.

bugadani, 2020-11-04:
> But do you think there would be a huge benefit of adding e-g only examples to the main e-g repo?

I think in the end, I'm just used to having the examples close to the source. But there are two points I see here:
* Ease of use. This is a less concrete point, but I think looking up an example for an older version of the source and browsing them together. I often do this, others may not, and with care an external repository can probably work as well.
* Maintainability. Separating the examples from the source means that the examples will always lag behind the source, while keeping them together has the advantage that the examples don't compile if they aren't updated together with the source. This can, on the one hand, be a bit annoying, but it also leads to always up to date examples, even in the development phase.
As I'm writing this, you've made a "2.5th" point over at e-t: providing examples for the master doesn't need extra work and can uncover issues. Otherwise, care must be taken with each release to make sure the examples depend on the released crate, etc.

> But all examples could use some improvements

It doesn't hurt if they worked :) I plan on releasing one more feature release before transitioning towards 0.7, and I think at that point I'll need to put some effort into pulling everything into shape.

rfuest, 2020-11-04:
> But all examples could use some improvements

Just to be clear, "all examples" referred to the e-g examples not the e-t ones. After reading my comment again I realized that it might have sounded rude and wasn't intended to be.

The example maintainability point is a very good one I hadn't considered before. Previously everything was contained in one repo and tested together to make sure the examples were always up to date. @jamwaffles how do you think we should make sure that the examples are up to date?

jamwaffles, 2020-11-07:
Sidenote: I've just moved jamwaffles/embedded-fonts to embedded-graphics/embedded-bdf. @rfuest please make a PR with your fork when you have a moment :). It would be great to get that crate into a usable state!

Examples are a tricky problem, mostly with maintenance and version sync as already discussed. The e-g-core split will probably make dealing with that easier (correct me if I'm wrong), but I'm still not sure what the right solution is. 

Once e-g-core is split out, presumably this would make it easier to have simulator based examples in e-g/e-g. I don't recall exactly what went wrong, but I've previous had issues using a `std` dev-dependency in a `no_std` project, so that's something to look out for.

Impl details aside, one way of organising these examples could be that `e-g/examples` always uses the latest release (stable or alpha, take your pick) and mentions so clearly in the readme. Larger examples that use multiple e-g crates or target real hardware could live here.

On the other side, e-g/e-g could house the current list of single file examples that are updated with master as e-g development progresses. This allows a quick browse of the `examples/` folder for flyby visitors and keeps it in sync with previous commits of the crate. I think, as long as the e-g-core version is the same for both e-g and simulator, this should work without introducing circular dependencies.

In the interim, should we just point the e-g/simulator examples to e-g/e-g master as a very hacky stopgap?

rfuest, 2020-11-08:
> Sidenote: I've just moved jamwaffles/embedded-fonts to embedded-graphics/embedded-bdf. @rfuest please make a PR with your fork when you have a moment :). It would be great to get that crate into a usable state!

Thanks, but it will probably take a while, because there are some other thing I want to work on first. And I will probably create a PR that only contains the changes in `bdf-parser` before merging the new code.

> Impl details aside, one way of organising these examples could be that `e-g/examples` always uses the latest release (stable or alpha, take your pick) and mentions so clearly in the readme. Larger examples that use multiple e-g crates or target real hardware could live here.

While having some examples in the main e-g repo would make them more discoverable, I still think that this would be the easiest solution for the dependencies. And we could have all example in a single place.

I would suggest the following development model for the `examples` repo. Each stable version of e-g gets a separate branch in the examples repo and the default branch is set to the branch for the latest stable version, because the average user is most likely interested in looking up examples for the latest stable release. When we start development of a new major version of e-g we create a new corresponding branch and keep that in sync with the changes in e-g. All links from the e-g docs to the examples repo will be extended to contain the branch name to make sure they always point to the appropriate example files. And as soon as we release we will change the default branch in the examples repo to the new stable branch.

We should also consider issue #421, which I had posted a while ago, and which describes a similar problem.

rfuest, 2020-11-10:
Another related issue, I have just noticed, are the font images in the docs. They are currently broken in the docs published on docs.rs: https://docs.rs/embedded-graphics/0.7.0-alpha.1/embedded_graphics/fonts/index.html#built-in-fonts because of the reorganization of the e-g repo. We should probably replace master by a tag or commit when we link from the docs to the repo.

rfuest, 2021-03-31:
@jamwaffles I believe this can be closed?

jamwaffles, 2021-04-01:
I think so!

closed 2021-04-01 by jamwaffles