embedded-graphics/embedded-graphics #460: Split core traits

jamwaffles, 2020-11-03:
This has been discussed before, IIRC with @rfuest raising the suggestion that we split core items like `DrawTarget` out into an `embedded-graphics-core` trait. 

I'm opening this issue because it might fix another issue: `tinytga` is used in the e-g doc examples, but `tinytga` depends on e-g itself, creating a cyclic dependency that breaks as soon as the versions aren't _perfectly_ in sync. This is obviously not good, so along with `DrawTarget`, we may want to consider splitting more traits for images and primitives, as well as the pixel colours, into a new core crate. 

If we go ahead with something like this, should it target 0.7 or go for a 0.8 release?

Discuss.

rfuest, 2020-11-03:
> I'm opening this issue because it might fix another issue: `tinytga` is used in the e-g doc examples, but `tinytga` depends on e-g itself, creating a cyclic dependency that breaks as soon as the versions aren't _perfectly_ in sync.

I think the easiest solution would be to remove any usage of `tinytga` in e-g itself. My suggestion would be to only use raw image in the e-g docs, which is enough to show the e-g side of the image API. In addition to that we could add links to some e-g doc examples that point to a copy of the example in the `tinytga` or `tinybmp` docs which uses a TGA or BMP file instead of raw data.

The `simulator` currently also has a dependency on `tinytga` and `tinybmp`, which caused some problems when I worked on e-f. But this shouldn't be a problem anymore if we split the examples into another repo.

> This is obviously not good, so along with `DrawTarget`, we may want to consider splitting more traits for images and primitives, as well as the pixel colours, into a new core crate.

I wouldn't want to add the entire `primitives` module to the core crate, because most of the types aren't necessary for the draw target or an image drawable. Instead we could only move the `Rectangle` struct, that is required by `DrawTarget` and `ImageDrawable`, to the core crate. Normal e-g users wouldn't notice this change because the `embedded-graphics` crate can reexport the `Rectangle` type in the `primitives` module.

I think the following types should be included in a core crate:
- `geometry::*`
- `pixelcolor::*`
- `draw_target::*`
  Not only `DrawTarget` is required, but also the other types, because they are used in `ImageDrawable` and the `tiny...` crates
- `image::ImageDrawable`
- `primitives::Rectangle` including all impls for traits that are moved into the core crate
  Other traits like `Primitive` should be implementable in `embedded-graphics`.

> If we go ahead with something like this, should it target 0.7 or go for a 0.8 release?

Maybe 0.7 would be a good time to implement this change, because it already requires bigger changes in the display drivers. But it isn't a priority IMO.

jamwaffles, 2020-11-03:
> I think the easiest solution would be to remove any usage of tinytga in e-g itself. My suggestion would be to only use raw image in the e-g docs, which is enough to show the e-g side of the image API.

Ok, that sounds like a good idea. For the current splitting work, I've made all the tinytga examples `rust,ignore` so we can focus on getting the crates split apart without having to worry about too much. After that we can fix the examples to remove the circular dependency.

> I think the following types should be included in a core crate:

Sounds good to me. I think you have permission to create repositories in the org, so if you'd like to start this, feel free to open `embedded-graphics-core` (or another name if you have a better one?)

> Maybe 0.7 would be a good time to implement this change, because it already requires bigger changes in the display drivers. But it isn't a priority IMO.

It isn't, but it would be good to get all these breaking changes out in one go - maybe then the next release won't require so much work to upgrade to ;)

rfuest, 2020-11-04:
> Sounds good to me. I think you have permission to create repositories in the org, so if you'd like to start this, feel free to open `embedded-graphics-core` (or another name if you have a better one?)

For a user of e-g the split into two crates won't be visible or relevant, because the main e-g crate reexports everything in the core crate. Because of this users will likely report any issues on the main e-g repo. For this reason I think it's better to leave the core crate inside the main repo, so we don't have to move issues to the relevant repo.

I don't want to work on this at the moment because I want to get the BDF support ready for 0.7, but I can look at this after that is finished.

jamwaffles, 2020-11-04:
Hm. Interesting point. I think I prefer the separate repo approach as releasing each crate is greatly simplified that way. In terms of users opening issues in the wrong repo, Github supports moving issues between repositories. Would that be enough to mitigate that problem IYO? It is IMO but there might be other issues you know of but haven't mentioned.

rfuest, 2020-11-05:
IMO it would be easier to maintain a single repo in general, but I cannot judge the additional effort that would require for the releases, maybe you can elaborate that a bit more.

Migrating issues between repos should be possible, but is an extra step and sometimes it might not immediately be clear which crate causes the issue. And sometimes an issue might even require changes in both crates.

Another advantage of having a single repo is that the latest version of e-g is always guaranteed to be in sync and this would be easily testable with CI. If an issue requires changes in both repos we would need to create separate PRs in both repos and merge them in order, because otherwise the e-g build would break if it depends on e-g-core changes.

jamwaffles, 2020-11-07:
I use [cargo-release](https://github.com/sunng87/cargo-release) to handle all the readme updates, tagging, etc, but it doesn't play nicely with workspaces, specifically `[patch.crates-io]` overrides in the root. It also struggles to update versions between crates in the same workspace.

That said, I don't want to let a release tool and it's issues dictate repo structure.

Another thought I just had, related to https://github.com/embedded-graphics/embedded-graphics/issues/455#issuecomment-723492027, is: would version management become more difficult if e-g-core is in the same workspace as e-g? I guess we could forego the workspace config and just have to separate crates in the same repo, but that doesn't feel great.

jamwaffles, 2020-11-07:
FWIW there seems to be an issue open to address the cargo-release problem: https://github.com/sunng87/cargo-release/issues/213. There are a few other similar issues in the backlog too IIRC.

rfuest, 2020-11-08:
> ...specifically `[patch.crates-io]` overrides in the root.

Maybe there are still other problems, but I don't think we need to use `patch`: https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#multiple-locations

How would we arrange the crates in the repo if we decide to use a single repo? Like we did before, with a dedicated workspace `Cargo.toml` in the root, or like https://github.com/BurntSushi/rust-csv where the main crate is in the root of the repo and the core crate is a subdirectory?

jamwaffles, 2020-11-09:
> Maybe there are still other problems, but I don't think we need to use patch

I didn't know that was a feature. Using dependencies like that could solve a lot of problems.

> How would we arrange the crates in the repo if we decide to use a single repo?

Preferably like we did before with a workspace `Cargo.toml`.

jamwaffles, 2020-11-10:
I haven't thought about this much, but how should we handle the `pub(crate)` items in the `geometry` module? E.g. `pub(crate) trait Trigonometry`.

rfuest, 2020-11-10:
I also haven't looked into this in detail, but IMO the cleanest solution would be to split the `geometry` module and only include `Point` and `Size` in the core crate. All types that aren't necessary for the image and draw target APIs can remain in the `embedded-graphics` crate. The remaining `pub(crate) fn`s in the `Size` type can either be made public or moved to an internal extension trait inside `embedded-graphics`.

jamwaffles, 2020-11-10:
Good point - there's no need to export all the angle stuff. There are only 4 `pub(crate)` items in `size.rs` and none in `point.rs` which makes that job easier:

```rust
// size.rs
pub(crate) fn saturating_add(self, other: Self) -> Self;
pub(crate) fn saturating_sub(self, other: Self) -> Self;
pub(crate) const fn from_bounding_box(corner_1: Point, corner_2: Point) -> Self;
pub(crate) fn center_offset(self) -> Self;
```

The `saturating_*` methods could be made pub (unsure why they weren't before), as could `from_bounding_box`. `center_offset` is a bit weird to make public so would be the only thing that would need to go into a `SizeExt` trait I think.

rfuest, 2020-11-10:
> The `saturating_*` methods could be made pub (unsure why they weren't before),...

I recently tried to use them in another crate and couldn't use them because they were private, so I agree that we should make them public.

> ..., as could `from_bounding_box`

The main reason for this being private, if I remember correctly, was that we couldn't find a good name for it. And I'm not sure how useful function is for most e-g users. The same function is available by using `Rectangle::with_corners(p1, p2).size` and perhaps this will even compile down to the same code when the compiler detects that `top_left` is never used. If this is the case the only advantage of `from_bounding_box` is that it can be `const` which `with_corners` can't be (at least in `1.40.0`).

> `center_offset` is a bit weird to make public so would be the only thing that would need to go into a `SizeExt` trait I think.

If we only have one function, that isn't used very often, I think we should just define it as a function and don't bother with an extension trait.

jamwaffles, 2020-11-10:
> The main reason for this being private, if I remember correctly, was that we couldn't find a good name for it.

That rings a bell, yes.

> If we only have one function, that isn't used very often, I think we should just define it as a function and don't bother with an extension trait.

I agree, although if we've got `from_bounding_box` too, that makes a slightly stronger case for an extension trait.

At any rate, this should be a reasonably clean split.

PR #498 (2020-11-29): Split core traits
Thank you for helping out with embedded-graphics development! Please:

- [x] Check that you've added passing tests and documentation
- [x] Add a `CHANGELOG.md` entry in the **Unreleased** section under the appropriate heading (**Added**, **Fixed**, etc) if your changes affect the **public API**
- [x] Run `rustfmt` on the project
- [x] Run `just build` (Linux/macOS only) and make sure it passes. If you use Windows, check that CI passes once you've opened the PR.

## PR description

This is going to be a horrific PR to review, so sorry for that in advance.

Do we want to re-exp [... cut]
closed 2020-11-29 by jamwaffles