embedded-graphics/embedded-graphics #461: Use baseline for text position

rfuest, 2020-11-03:
Originally posted in #455:

The position field in Text objects currently specifies the top left corner of the texts bounding box. This can cause problems with other text renderers, where each glyph can have a different size and where the bounding box height depends on the actual text. For this reason other APIs use the left most point on the baseline to define the text position. This also makes it easier to mix different fonts, font sizes and even renderers on a single line of text.

To implement this change for the internal font renderer I would like to add a BASELINE constant to the Font trait, which defines the offset between the baseline and the top of glyph bounding box.

jamwaffles, 2020-11-07:
This sounds like a very good idea, but I think some of the fonts we already use have different baselines for different characters. For example, `Font12x16` has `p` and `q` above their baseline relative to other chars:

[image]

I'm thinking the `BASELINE` property would be more useful for the BDF fonts, but I'd like to clarify how it works for the case shown above.

rfuest, 2020-11-08:
I wouldn't consider the baseline of `p` and `q` in `Font6x8`, and its derivatives, different than the other glyphs. The top of the glyph aligns with all the other lower case letters and the rounded part starts above the baseline.

But regardless if this fonts uses different baselines per glyph or not, IMO we shouldn't complicate the internal font renderer by adding support for different baselines per glyph. For `Font6x8` the baseline would be set so that the lower part of the `s` would be on the baseline and `p` and `q` would extend 1 px below the baseline.

jamwaffles, 2020-11-08:
That all makes sense, thanks for explaining. Looking at my screenshot again I notice `p` and `q` are indeed offset a bit.

> IMO we shouldn't complicate the internal font renderer by adding support for different baselines per glyph. 

Agreed :)

rfuest, 2020-11-16:
If we change this anyway we could also consider to add a few more options, similar to what was suggested in: https://github.com/embedded-graphics/embedded-graphics/issues/225 Sometimes it's much easier to correctly position a text with center horizontal alignment, for example in the `analog-clock` example (not considering the additional background).

Instead of one enum, as suggested in #225, I would suggest to add separate horizontal and vertical ones:
```rust
enum HorizontalAlignment {
    Left, // <- default
    Center,
    Right,
}

enum VerticalAlignment {
    Top,
    Center,
    Bottom,
    Baseline,  // <- default
}
```

jamwaffles, 2020-11-16:
Makes more sense than the giant enum I proposed :)

closed 2020-12-14 by rfuest