embedded-graphics/embedded-graphics #462: oc/ folder management

jamwaffles, 2020-11-07:
- Version of embedded-graphics in use (if applicable): [version here]
- Target: [simulator] / [real hardware]
- If using real hardware: [display driver used]

Please link to the driver on <https://crates.io> if applicable.

## Description of the problem/feature request/other

[description here]

## Test case (if applicable)

```rust
// Failing test case demonstrating the issue here
```

rfuest, 2020-12-07:
I just ran into this issue when I tried to update the `simulator` dependencies. The `generate-drawing-examples` binary requires a dependency to `tinytga`, so that the simulator crate must have a `dependency` on `tinytga` and cannot use a `dev-dependency`.

Should we move the code for the `generate-drawing-examples` binary into the `build-tools` repo? If we do that we would need to also clone the `build-tools` repo during builds of this repo, to check if the examples in the doc folder are up to date.

jamwaffles, 2020-12-11:
It would make sense to move them into `build-tools`, yes. It's a public repository so I don't think cloning it in CI is a problem. (I've had lots of issues in the past cloning private repos)

rfuest, 2020-12-14:
Another option would be to add a separate crate for the examples generator to the `embedded-graphics` repo, that is not part of the workspace. IMO this would be better, because we could keep everything in one repo and it shouldn't mess with any dependencies when it's not even part of the workspace.

We should also change the tool to output markdown directly, instead of having to use `sed` in the `justfile`. This would bring us one step closer to being able to run `just build` on Windows.

closed 2021-02-05 by rfuest